nielsonm236/NetMod-ServerApp #1: Server not online

bukum86, 2020-08-15:
Hello, 1st of all i would like to thank you for effort you made reprograming this useful server. I followed your guide and run into problem. I would try to explain. Server comes online only for few seconds and than its unreachable. I have the feeling that its working very slow and that its going into some kind of loop. One thing i noticed while loading project into STVP is that i get this error no matter what version of code I use (16in, 16out, 8out). 

What might be the problem? 

FILE : line 952: Address 0x4000 is out of range and is ignored!
FILE : line 952: Address 0x4001 is out of range and is ignored!
FILE : line 952: Address 0x4002 is out of range and is ignored!
FILE : line 952: Address 0x4003 is out of range and is ignored!
FILE : line 952: Address 0x4004 is out of range and is ignored!
FILE : line 952: Address 0x4005 is out of range and is ignored!
FILE : line 952: Address 0x4006 is out of range and is ignored!
FILE : line 952: Address 0x4007 is out of range and is ignored!
FILE : line 952: Address 0x4008 is out of range and is ignored!
FILE : line 952: Address 0x4009 is out of range and is ignored!
FILE : line 952: Address 0x400A is out of range and is ignored!
FILE : line 952: Address 0x400B is out of range and is ignored!
FILE : line 952: Address 0x400C is out of range and is ignored!
FILE : line 952: Address 0x400D is out of range and is ignored!
FILE : line 952: Address 0x400E is out of range and is ignored!
FILE : line 952: Address 0x400F is out of range and is ignored!
FILE : line 952: Address 0x4010 is out of range and is ignored!
FILE : line 952: Address 0x4011 is out of range and is ignored!
FILE : line 952: Address 0x4012 is out of range and is ignored!
FILE : line 952: Address 0x4013 is out of range and is ignored!
FILE : line 952: Address 0x4014 is out of range and is ignored!
FILE : line 952: Address 0x4015 is out of range and is ignored!
FILE : line 952: Address 0x4016 is out of range and is ignored!
FILE : line 952: Address 0x4017 is out of range and is ignored!
FILE : line 952: Address 0x4018 is out of range and is ignored!
FILE : line 952: Address 0x4019 is out of range and is ignored!
FILE : line 952: Address 0x401A is out of range and is ignored!
FILE : line 952: Address 0x401B is out of range and is ignored!
FILE : line 952: Address 0x401C is out of range and is ignored!
FILE : line 952: Address 0x401D is out of range and is ignored!
FILE : line 952: Address 0x401E is out of range and is ignored!
FILE : line 952: Address 0x401F is out of range and is ignored!
FILE : line 953: Address 0x4020 is out of range and is ignored!
FILE : line 953: Address 0x4021 is out of range and is ignored!
FILE : line 953: Address 0x4022 is out of range and is ignored!
FILE : line 953: Address 0x4023 is out of range and is ignored!
FILE : line 953: Address 0x4024 is out of range and is ignored!
FILE : line 953: Address 0x4025 is out of range and is ignored!
FILE : line 953: Address 0x4026 is out of range and is ignored!
FILE : line 953: Address 0x4027 is out of range and is ignored!
FILE : line 953: Address 0x4028 is out of range and is ignored!
FILE : line 953: Address 0x4029 is out of range and is ignored!
FILE : line 953: Address 0x402A is out of range and is ignored!
FILE : line 953: Address 0x402B is out of range and is ignored!
FILE : line 953: Address 0x402C is out of range and is ignored!
FILE : line 953: Address 0x402D is out of range and is ignored!
FILE : line 953: Address 0x402E is out of range and is ignored!
FILE : line 954: Address 0xA is out of range and is ignored!
FILE : line 954: Address 0xB is out of range and is ignored!
FILE : line 954: Address 0xC is out of range and is ignored!
FILE : line 954: Address 0xD is out of range and is ignored!
FILE : line 954: Address 0xE is out of range and is ignored!
FILE : line 954: Address 0xF is out of range and is ignored!

nielsonm236, 2020-08-15:
Sorry for the delay in responding. This is my first GitHub project so I'm still learning how it works.

I get those same out of range messages, but they aren't errors. I could dig deeper but I believe it is just indicating that those items are out of range of the Flash memory area that is being programmed (which is true). So, they are not a problem. My devices program and operate normally.

My server modules have been operating for about 2 months without going offline. Admittedly I don't access them often and only use them for simple relay on/off functions. Are you trying to drive the relays rapidly? If that is the case then it might be possible that the server simply can't process the incoming commands fast enough.

Mike

nielsonm236, 2020-08-15:
One other thought: When I first hooked up 16 relays my power supply was not adequate to handle the surge current when the relays switched. This caused power to the network module to dip low enough to reset the module. If you have relays connected (particularly if there are a lot of relays) you may want to check if the power supply is adequate.

nielsonm236, 2020-08-15:
I checked my notes with regard to the out of range messages. The 0x4000 to 0x407f items are in the EEPROM, and the 0x0000 to 0x5ff are in RAM. The programmer can't write to these locations, so any references to them are considered "out of range". I understood this to be a warning just in case I was intending to write to them (for instance, if I had tried to initialize RAM or EEPROM with the programmer). I'm not doing that so I can ignore the warnings. All EEPROM and RAM read/writes are via runtime code.

nielsonm236, 2020-08-15:
If the above suggestions don't help with keeping the Network Module online let me know some more about your configuration. As FYI most of my testing has been within a local test network (no attempt to access over the internet). There shouldn't be any issue with internet access with the proper port forwarding in your router, but I haven't tried it yet. So if there is still a problem let me know how you've got things configured and I will try it on my test network.

bukum86, 2020-08-15:
Hi Mike, you did really good work writting detaild guide. I was wondering if those error messages are part of problem i have. I havent connected any relays to the module yet. I used exactly same hardware in your guide. after successful flashing i am not able to access server trough local network. After I reset module several times i was able to get access to front page. After few seconds i try to refresh it and i saw that page is no longer accessable. Next thing i can try is to connect module directly to my desktop via LAN cable and see if it will work, but i think that problem is not in my local network

nielsonm236, 2020-08-15:
This is very odd. I don't think the messages have anything to do with it. I see the same messages when I program a module, and I do not have an issue. Is this the only module on your network? I'm asking because I wonder if there is more than one device with the same MAC address. If so, the router will get very confused and access to the device will be intermittent. Also I recommend that you leave the most significant byte of the MAC address at C2 (the "2" being the important part). I think doing a direct connect to your PC/Laptop may tell you more.

nielsonm236, 2020-08-16:
I had another thought that might be helpful. From your comments it sounded like you might have done your initial programming with the Network Module connected to your network rather than directly connected to your PC/Laptop. That can be done, but be aware that your router or switch will associate IP addresses and MAC addresses in its ARP table, and depending on the router it may also track which physical port of the router/switch a MAC/IP address pair appears on. Because of this some routers/switches get confused for a short time when MAC/IP address pairs change. This is why it is usually safer to do all the Network Module programming and MAC settings when connected directly to a PC/Laptop.

Still, it *should* be OK to make these changes connected to your network but it may take your routers/switches up to a minute or more to register the changes. Until the changes are registered in the ARP tables the device may be inaccessible. Even a direct connect to a laptop takes a few seconds for the OS socket to adjust to such changes. You see that happen to me in the "Programming" video when I made changes, then I comment that it might not have worked - and then it started working. In my network changes like this adjust in about 10 to 15 seconds, but it may take longer with different routers/switches. Of course the most important thing is that no MAC address appears on more than one device.

So far I have not had to reset ARP tables in any routers or switches as a result of doing MAC changes, but that is a possible way to clear confusion that occurred. A power cycle of the router or switch would do this. But I don't think that is necessary.  Sometimes after changing a MAC address just moving the device connection to a different physical port on the switch will cause the ARP tables to update. How complex is your network? Just one gateway router at your Internet connection? Or do you have one or more switches or additional routers layered behind your gateway? I'm just trying to think this through as I'm not hearing from anyone else with this issue.
Thanks, Mike

bukum86, 2020-08-16:
Hi Mike, thank you for thinking about my problem. My home network is very simple. There is only cable internet router with 4 LAN slots. All devices are connected to internet via Wifi. I just plugged in new WEB_Relay with its original software and its recognized on the network immediately. I tried with restarting router and changing of LAN slots. Nothing helps. By the way, is there possibility to download original software of my web_relay so i can flash it back if somethings is not working in the end? My problem at the moment is that I dont have desktop here, and laptop dont LAN port. WEB_relay and PC direct connection is not possible at the moment.

bukum86, 2020-08-16:
Good news. I tried on another unit of WEB_Relay and it works perfectly, as it should be. I am still puzzled what happened with 1st unit. I noticed that one cheap was very hot (maybe high temperature destroy or damaged chip?). On second one I added small aluminium cooling block. By the are there commands (like in standard version, 192.168.1.100/30000/01 etc) which we can use to control relays?

bukum86, 2020-08-17:
I managed to successfully flash 3 units. 1st one is probably damaged by me or it left factory with hidden mistake. It puzzled me that it sometimes shows page and it goes offline afterwords but well this is the life :)

Thanks Mike for your support.

nielsonm236, 2020-08-17:
Answering in order of your comments:
- The original firmware is not available, which is why I wrote my own. I tried very hard to get the supplier to provide code (even the binary) thinking I could start from there to develop fixes, but there is no support at all from them. They just run you in circles providing non-answers. As stated in my documentation and videos once you erase the device the only way to make it work again is to program it with the code I supplied.

- It is possible the first unit was defective from the factory. These are fairly simple so close examination might reveal a bad solder connection you could fix. Another user reported that they also have one that does not seem to work at all - but several more that are working fine. If you have a chip that is very hot that would explain why it works for a few seconds then stops. Once it gets hot enough it fails. I have 5 of the devices on my network right now and they get warm but not hot. I was wondering why they supplied a heat sink as mine are not getting hot enough to need it. There was no documentation showing me which chip they wanted the heat sink on so I was waiting to see if a chip got warm enough to give me a clue.

- With regard to REST types of commands, look at the documentation on GitHub or the Help web pages on the device (click on the Help button when the Network Module web pages appear). There are REST type commands to operate all relays and several other functions. For instance a command like 192.168.1.100:8080/00 will turn relay 1 off, and 192.168.1.100:8080/01 will turn relay 1 on. Note that I implemented actual port numbers, so there is a ":" between the IP Address and Port number, not a "/". If you set the devices to use port 80 you do not need the ":80".

I am happy that you have some modules working now. I was very puzzled about the problem you described. With all the work you had to do you are close to being an expert flashing the devices by now. I hope you find the firmware useful. I am working on an MQTT interface that should make the devices more useful in home automation (perhaps even Alexa interfacing).

Let me know of any issues or suggestions for changes. The Flash and RAM space is very limited, but I might be able to add improvements if they will fit.

Mike

nielsonm236, 2020-08-17:
If you can recall let me know which chip was getting hot. The one with pins on all four sides? Or the one with pins on two sides?

bukum86, 2020-08-17:
> If you can recall let me know which chip was getting hot. The one with pins on all four sides? Or the one with pins on two sides?

Hi Mike, Chip with pins on two sides is getting hot in my case. 

Thanks for info about 192.168.1.100:8080/00. I want to control relays from linux command line with curl command. 

In the mean time i made another try using NodeMCU ESP8266 to control 8 relays with Blynk app. Its also nice plug and play solution. Next try will be with ESP22 and 16 relay board. This is very nice solution because relays can be controlled directly from my phone anywhere in the world. Not just from local network. 

Sorry for this off topic. Thank you very much again for your help and awesome work.

nielsonm236, 2020-08-17:
Thank you for the info on the hot chip - the one you identified is the ENC28J60. The chip itself is very reliable, but the module construction might have an error in it. I will look at my devices more closely to see if there is something fragile that I should let you know about. My initial thought is to look for debris or a bent pin in the RJ45 connector - some short or "almost short" in there could make the ENC28J60 pin drivers overheat.

Thank you for your experience info on the ESP8266 and ESP22. My main reason for being interested in the Web_Relay_Con device is that it is wireless. I use them in security power control, and in my location wifi is very congested.  Too many wifi routers too close together, so lots of interference causing problems with low power wifi devices, and I needed very reliable connectivity to the devices. But many people won't see that interference so wifi devices will work well.

I'll close this issue, but stay in touch if you have more comments. And I will put information out as I get MQTT working and go on to developing tools for internet access to the modules. You can subscribe to GitHub or the YouTube videos and should get notice of those updates.

Thank you for your help on resolving this issue. The conversation may be helpful to others.

Mike

closed 2020-08-17 by nielsonm236