nielsonm236/NetMod-ServerApp #6: Changing output states over the REST interface

dom2049, 2020-11-28:
Hi,

Changed my IP to 192.168.1.123 and trying to change states of one relay output. It changes fine when using your web interface, but when calling http://192.168.1.123:8080/00 and http://192.168.1.123:8080/01, your web IO status shows the change, but the relay doesn't change state? I have to click "Save" on the IO config page.

nielsonm236, 2020-11-28:
I will retest this. If you use the REST commands you should see the relay output change immediately, but the IOControl page won't show the change until you click "Refresh". I will get back to you shortly. Can you tell me what revision shows on the Configuration page?

dom2049, 2020-11-28:
It' showing Code Revision 20201126 1527.

nielsonm236, 2020-11-28:
OK - That is the latest revision, and I just reproduced the problem. I will get to work on it right now.

nielsonm236, 2020-11-28:
Just fixed it with release 20201128 1733. I will get this uploaded within the next few minutes. The issue was sourced by a recent change for MQTT support and the REST commands did not get re-tested after that. Fix was simple so it should not cause any other issues. Thank you for finding this. If you notice anything else let me know!

closed 2020-11-28 by nielsonm236