nielsonm236/NetMod-ServerApp #8: REST IO status page 99 should not format the output as HTML

dom2049, 2020-11-28:
The REST status page at http://192.168.1.123:8080/99? should be 16 bits long and not have all the extra HTML stuff that needs to be parsed and filtered out as it is not meant for human reading anyway. Just a suggestion.

nielsonm236, 2020-11-28:
I am not quite sure what is being requested, and perhaps there is something for me to learn here. The incoming request (" http://192.168.1.123:8080/99?") is in the form of an html request, and the response is in kind. I can strip away the html ... but to meet the http specs I would have to at least wrap the reply in some protocol - one the requesting source wouldn't understand if the source is html based. The MQTT implementation is probably the smallest http request/reply implementation, and does have a command implemented to reply with a two byte binary response for the IO states.

Explain further what you mean with regard to the /99 html request/response and I will look into it. Right now it is implemented to mimic the response provided by the original factory code (not that I see it as particularly useful ... except to those that did some development with the factory code).

dom2049, 2020-11-28:
I know you're satisfying the HTML standard by outputting the HTML and BODY tags, but for this purpose it is really not necessary.  I just need to do this below to parse the output and get the 16 bits and it is ugly.
res.Substring(res.IndexOf("<body><p>") + 9, 16);
Maybe you could add an extra status page such as 999 or 98 that would output just the 16 status bits, without the HTML tags.

dom2049, 2020-11-28:
99 source is 

  | <!DOCTYPE html><html lang='en-US'><head><title>Help Page 2</title><link rel='icon' href='data:,'></head><body><p>0001111111111111</p></body></html>
-- | --
 
when all there is needed is 0001111111111111

nielsonm236, 2020-11-29:
I will experiment with creating a TCP payload as you describe to see what happens. Do you suggest that it be 16 alphanumeric characters? Or an actual number 0 or 1? Or 2 bytes with the bits representing the data?

nielsonm236, 2020-11-29:
dom2049 - Send me an email at nielsonm.projects@gmail.com. I'll send you a test build to try out using 16 alphanumeric characters and no html formatting. Which build type are you using? 16out, 8out, 16in, or 8outMQTT?

nielsonm236, 2020-11-30:
Update: I sent dom2049 a test code and he reports that it provides the behavior he was looking for (via command /98). While documenting the change I found a typo on the Help pages which will also be corrected.

nielsonm236, 2020-11-30:
Addressed in Code Revision 20201130 1752

closed 2020-11-30 by nielsonm236