nielsonm236/NetMod-ServerApp #11: All Relays turns ON and OFF Power ON And Restart

aryehassbian, 2020-12-02:
Hi,
I have downloaded the latest version of 16Outputs and connected the device to an Eight Relay Module.
When I turn the module it switches all the relays ON and then back OFF. Also my relays are all active low that means when I turn the relay ON it will turned OFF and vice versa.
The older version had an invert - I could not find it in the new version.
I need to get rid of the turn on of all relays at start and also to be able to invert the states of the relays.
can a flag be set in the EEPROM to say if the Relay is Active High or Low and By reading it it will set the interface and pull the relays to their previous state on Boot and Restart?

Thanks!

nielsonm236, 2020-12-04:
I have to take back a little of what I said with regard to #3 above. If everything remains powered up and all that happens is a reboot I was able to change the code to at least give the appearance that all relay states are maintained (through the reboot). I say "gives the appearance" because a reboot, being a hardware reset, turns the output pins into "floating input pins" for a very brief period. My code change pre-writes the Output Data Registers early in the processor restart, then changes the registers from their default reset state (input) to outputs. This apparently happens fast enough that relays don't change state.

As FYI I'm adding a section to the manual to describe the necessary external hardware needed if you want all relays to maintain their states even in the event of total power loss. The hardware to do that is simple in concept, but difficult in terms of implementation as enough components are involved that a circuit board needs to be made to support latching relays. For most applications a UPS is the better solution (ie, don't lose power). Another user I'm working with uses POE switches and powers the Network Module and Relays via POE with a 48V-to-5V converter. In that configuration a UPS is only needed at the POE switch.

nielsonm236, 2020-12-04:
Code Revision 20201204 1506 fixes the reboot chatter issue. "Complete power loss" handling suggestions are now described in the documentation. Closed.

closed 2020-12-04 by nielsonm236