nielsonm236/NetMod-ServerApp #14: Renaming buttons

s57aw, 2020-12-12:
Thanks for a great project, works very well and stable for me.

s57aw, 2020-12-12:
Thanks a lot for letting me now.... I suspected something like this, lack of EEPROM storage to retain the names. But no problem, it is not big issue at all, not so important and I understand that this small and cheap device has its limits :-)

Thanks again for all your work

s57aw, 2020-12-12:
> Hi Robert - I wanted this too ... but the problem is lack of EEPROM
> storage to retain the names through reboot and power loss. The EEPROM is
> limited to 128 bytes, and the code currently uses 82 of those. At least 10
> bytes need to be reserved for future features. That would leave 36 bytes,
> and 16 things to name, so there would be only 2 characters per name. So I
> gave up on that idea and I'm keeping the balance of the EEPROM for debug.
> HOWEVER - I believe Flash can also be written by code - opening the
> possibility that this feature could be added. Flash has a much more limited
> write cycle count spec (100 cycles vs EEPROM at 100,000 cycles). Those
> specs are at worst case conditions, so the typical cycle count would be
> much larger, perhaps multiple times larger.
> I will keep this open a while to remind me to look into Flash as
> non-volatile storage. I seem to recall there are some caveats that make
> this harder to do than EEPROM, but worth another look.

I am just wondering if you can "save" some storage / get space for that if
you program only 8 outputs ignoring 9 - 16 outputs/inputs. For example, I
am using 8 channel Arduino relay board with HW-584 and 8 relays are more
than enough for me, I do not need additional 8 outputs or inputs.

Thanks & stay well

Robert
Ljubljana, Slovenia

nielsonm236, 2020-12-12:
That approach only saves 1 byte in EEPROM. The pin states are stored 8 pins per byte. The EEPROM map is shown in the reprogramming document on GitHub so you can see what is there. I need to keep all EEPROM bytes located where they are so that people don't lose their settings when they upgrade firmware or change build types (say, changing from non-MQTT to MQTT).
I really think the way to make this work is to store names in Flash, so I need to re-study that.

nielsonm236, 2021-02-28:
Update: I've reviewed the STM8S specs again, and I do see how to save very infrequently changing user inputs to Flash instead of EEPROM. So from a technical perspective this request could be addressed. The remaining question is how to do this if MQTT code is also compiled in the build. It looks like managing variable IO names would require more code than will fit along side MQTT code, and for this specific feature I'm reluctant to have this aspect be different in the GUI for "MQTT" vs "Browser Only" builds.
Note to self: I suggest using 16 character IO names. This would occupy 256 bytes of Flash for the names. That is big, but the bigger part would be the javascript, parsing, and management of the names in code. Fits easily in a "Browser Only" build, but won't fit in an MQTT build (at least not at this date). If we could only find another 500 to 750 bytes of space ...

nielsonm236, 2021-03-08:
Update for anyone following this issue: We are implementing this concept now. There is still the question of whether this will fit in our current approach of having a single build. This will easily take another couple of weeks to know.

nielsonm236, 2021-04-01:
I'm in the final steps of testing this feature for release. I need to add documentation so I'm guessing about a week to go.
Adding this feature required that the firmware get split into separate releases again due to Flash space limitations. One release will the "For MQTT users" and one is "For Browser Only users". The IO Names feature will be in the Browser Only firmware. Since MQTT management devices provide plenty of methods to do this kind of IO Naming I think it is appropriate that it be split this way ... that plus the fact that it wouldn't fit in Flash with the MQTT code.

nielsonm236, 2021-04-12:
Released with 20210412.1333

closed 2021-04-12 by nielsonm236