nielsonm236/NetMod-ServerApp #15: MQTT Improvements

yozik04, 2020-12-16:
Thank you for spending your time on this great firmware. I have not tried it so far but I am about to start.

Inspecting MQTT topics and see that publishing part is very inconvenient. It would be way better if you'd publish this way:

### Getting state
Publish for every change
`NetworkModule/<devicename>/pin/<input #>`, payload `ON` or `OFF`
Same with relay status:
`NetworkModule/<devicename>/pin/<output #>`, payload `ON` or `OFF`
Both retained.

### Setting state
`NetworkModule/<devicename>/pin/<output #>/set`, payload `ON` or `OFF`

`all` could be set this way:
`NetworkModule/<devicename>/pin/all/set`, payload `ON` or `OFF`

### Last will
It would be also great to have a last will:
`NetworkModule/<devicename>/availability`, payload `offline`
Retained

When device boots it would publish
`NetworkModule/<devicename>/availability`, payload `online` and set the last will. Would be good to republish input and output statuses as well. All retained.

Retainability could be configurable, but if you would not retain then there should be a republish interval. Extra effort that has a really low benefit in practice.

This is standard for HomeAssistant.
https://www.home-assistant.io/integrations/switch.mqtt/

Next steps would be to add publishing of HomeAssistant MQTT Autodiscovery configuration. Of course if flash size will allow. This way your firmware will find A LOT of usage in HomeAssistant community. Imagine people just configure MAC Address, IP, MQTT settings and they instantly get the device with all switches in HomeAssistant with zero configuration on HomeAssistant side.

I can help with coding. =)

nielsonm236, 2020-12-16:
Hi - I do have a last will implemented per the document (search on Network Module Publish Status). I will look over the alternative messaging you suggest here - I already have some users implementing with the current messaging, but can probably support both in parallel. It is always a struggle between memory space and features, but at the same time I know there is still a good bit of inefficiency in the code due to my own methods and re-use of code from others. So, may still be able to fit it in.

yozik04, 2020-12-16:
Looked in the code and found last will code. Great! One step less.

Supporting two in parallel is not a great idea. I'd release a new major version that has a more standard approach. But it is up to you do decide.

nielsonm236, 2020-12-17:
After some discussion today with a fellow developer (Carlos in Portugal) I think your suggestion is a good idea. I'm not sure how to get it communicated to other users, but all will need to change their MQTT commands. Thank you for the suggestion and pointers - this is my first MQTT project so the help is much appreciated.

yozik04, 2020-12-17:
There are always a lot of changes in open source software. Just keep changelog updated and have old version available. The change won’t be hard for the end users. I have two MQTT projects in my background. I think I have described all core principles in my first email.

nielsonm236, 2020-12-17:
Jevgeni - I propose the following at the entire set of Subscribe and Publish messages. Please pay special attention to the "availability" messages to make sure I understood the Home Assistant documentation correctly. Let me know if this looks complete. If yes, I will start making the code changes and verify that it all fits OK.

**Publish sent to Network Module to cause a change:**
NetworkModule/<devicename>/pin/output1/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output2/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output3/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output4/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output5/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output6/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output7/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output8/set Payload: "ON" or "OFF"

NetworkModule/<devicename>/pin/all/set Payload: "ON" or "OFF"

**Publish sent to Network Module to request binary pin state information:**
NetworkModule/<devicename>/pin/state-req

**Published sent to Network Module to request availability state:**
  NetworkModule/<devicename>/availability
  Note the module can only respond with "online", as it is
  incapable of responding with "offline". See the last will
  message below.

**Published by Network Module when a pin state change occurs:**
NetworkModule/<devicename>/pin/input1 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/input2 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/input3 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/input4 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/input5 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/input6 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/input7 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/input8 Payload: "ON" or "OFF"

NetworkModule/<devicename>/pin/output1 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output2 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output3 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output4 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output5 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output6 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output7 Payload: "ON" or "OFF"
NetworkModule/<devicename>/pin/output8 Payload: "ON" or "OFF"

**Published by Network Module in response to receiving
  "NetworkModule/<devicename>/availability" if module is online:**
NetworkModule/<devicename>/availability Payload: online

**NetworkModule/<devicename>/state payload 2 byte binary pin values**
  This message is only published in response to a “state-req” topic
  message. The payload consists of two bytes with the bits organized
  as follows:
  First byte:
    Bit 7 = Pin 16 (Input 8) state
    Bit 6 = Pin 15 (Input 7) state
    Bit 5 = Pin 14 (Input 6) state
    Bit 4 = Pin 13 (Input 5) state
    Bit 3 = Pin 12 (Input 4) state
    Bit 2 = Pin 11 (Input 3) state
    Bit 1 = Pin 10 (Input 2) state
    Bit 0 = Pin 9   (Input 1) state
  Second byte:
    Bit 7 = Pin 8   (Relay 8) state
    Bit 6 = Pin 7   (Relay 7) state
    Bit 5 = Pin 6   (Relay 6) state
    Bit 4 = Pin 5   (Relay 5) state
    Bit 3 = Pin 4   (Relay 4) state
    Bit 2 = Pin 3   (Relay 3) state
    Bit 1 = Pin 2   (Relay 2) state
    Bit 0 = Pin 1   (Relay 1) state

**Any client using the Network Module can Subscribe to
  “NetworkModule/<devicename>/availability” to receive the
  Network module online / offline messages.**

**When the Network Module connects to the Broker it will Subscribe
  to the following topics:**
NetworkModule/<devicename>/pin/+/set
  The above causes the broker to send the module the "set"
  commands, but won't reflect the modules own "pin" state
  messages back to the module (reduces traffic).

NetworkModule/<devicename>/pin/state-req

**When the Network Module connects to the Broker it will establish
  a "last will" message of "offline" with the will topic
  “NetworkModule/<devicename>/availability”**

**When the Network Module connects to the Broker it will Publish
  to the following topics:**
  NetworkModule/<devicename>/availability Payload: online

yozik04, 2020-12-17:
Let's save some memory and ease parsing:
```
NetworkModule/<device>/<object_type>/<object_id>[/set]
```

### Publish sent to Network Module to cause a change:
```
NetworkModule/<device>/output/1/set Payload: "ON" or "OFF"
NetworkModule/<device>/output/2/set Payload: "ON" or "OFF"
...
NetworkModule/<device>/output/all/set Payload: "ON" or "OFF"
```

You should subscribe to 
```
NetworkModule/<device>/output/+/set Payload: "ON" or "OFF"
```

`+` is a wildcard in MQTT

```
strcat(topic_base, "/output/+/set");
```

### Publish sent to Network Module to request binary pin state information:
> NetworkModule/<device>/output/state-req

This conflicts with output states topics. Can be
```
NetworkModule/<device>/state-req
```

But this won't be needed if MQTT retains state and publishes state after boot.

#### Response
``` 
NetworkModule/<device>/state payload 2 byte binary pin values
```
Same as above. If you want it. It is duplication of functionality.

-----------------------------------------------
### Availability

> Published sent to Network Module to request availability state:
NetworkModule/<device>/availability

Other way around. HomeAssistant does not send anything to query availability data. This is topic for device to notify HomeAssistant if it went online or offline. It works with last will.
When Network Module boots and connects to MQTT the first message it sends is:
```
NetworkModule/<device>/availability "online"
```
During connect it sets the last_will to publish to
```
NetworkModule/availability "offline"
```
It will auto trigger only when MQTT Broker will detect that connection with client was lost. Network error or device lost power, hanged or just does not send keep alive...

If module wants to disconnect intentionally it should send
```
NetworkModule/<device>/availability "offline"
```
Because last will will not trigger

--------------------------------

### Published by Network Module when a pin state change occurs and after boot:
#### Input
```
NetworkModule/<device>/input/1 Payload: "ON" or "OFF"
NetworkModule/<device>/input/2 Payload: "ON" or "OFF"
...
```

#### Output
```
NetworkModule/<device>/output/1 Payload: "ON" or "OFF"
NetworkModule/<device>/output/2 Payload: "ON" or "OFF"
...
```

Off topic: I think I will try to add DS18B20 support in future. Because I will need it. Planning to control heating valves.

### When the Network Module connects to the Broker it will Subscribe to the following topics:
```
NetworkModule/<device>/output/+/set
```

> The above causes the broker to send the module the "set"
commands, but won't reflect the modules own "pin" state
messages back to the module (reduces traffic).

True

Tip: To post code chunks wrap them with 3 backticks
\`\`\`
code
\`\`\`

nielsonm236, 2020-12-17:
I updated my notes to show what I think you described above. Two comments:
a) The state-req and state messages were at the request of another user that wanted a single command for a Note-Red interface that would return all the pin states in as few bytes as possible. The state message returns two bytes containing all the pin states in a binary format.
b) With the new definition a client that subscribes to "NetworkModule/<devicename>/input/#" to receive all input changes, and subscribes to "NetworkModule/<devicename>/output/#" to receive all output changes will also see all their "set" commands reflected back to their client (for instance "NetworkModule/<devicename>/output/1/set"). I don't know if this is an issue or not, but stopping this from happening is part fo the reason the current message set was defined the way it is.

Here's the updated definition:
**Publish sent to Network Module to cause a change:**
NetworkModule/<devicename>/output/1/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/2/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/3/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/4/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/5/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/6/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/7/set Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/8/set Payload: "ON" or "OFF"

NetworkModule/<devicename>/output/all/set Payload: "ON" or "OFF"

**Publish sent to Network Module to request binary pin state information:**
NetworkModule/<devicename>/state-req

**Published by Network Module when a pin state change occurs:**
NetworkModule/<devicename>/input/1 Payload: "ON" or "OFF"
NetworkModule/<devicename>/input/2 Payload: "ON" or "OFF"
NetworkModule/<devicename>/input/3 Payload: "ON" or "OFF"
NetworkModule/<devicename>/input/4 Payload: "ON" or "OFF"
NetworkModule/<devicename>/input/5 Payload: "ON" or "OFF"
NetworkModule/<devicename>/input/6 Payload: "ON" or "OFF"
NetworkModule/<devicename>/input/7 Payload: "ON" or "OFF"
NetworkModule/<devicename>/input/8 Payload: "ON" or "OFF"

NetworkModule/<devicename>/output/1 Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/2 Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/3 Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/4 Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/5 Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/6 Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/7 Payload: "ON" or "OFF"
NetworkModule/<devicename>/output/8 Payload: "ON" or "OFF"

**NetworkModule/<devicename>/state payload 2 byte binary pin values**
  This message is only published in response to a “state-req” topic
  message. The payload consists of two bytes with the bits organized
  as follows:
  First byte:
    Bit 7 = Pin 16 (Input 8) state
    Bit 6 = Pin 15 (Input 7) state
    Bit 5 = Pin 14 (Input 6) state
    Bit 4 = Pin 13 (Input 5) state
    Bit 3 = Pin 12 (Input 4) state
    Bit 2 = Pin 11 (Input 3) state
    Bit 1 = Pin 10 (Input 2) state
    Bit 0 = Pin 9   (Input 1) state
  Second byte:
    Bit 7 = Pin 8   (Relay 8) state
    Bit 6 = Pin 7   (Relay 7) state
    Bit 5 = Pin 6   (Relay 6) state
    Bit 4 = Pin 5   (Relay 5) state
    Bit 3 = Pin 4   (Relay 4) state
    Bit 2 = Pin 3   (Relay 3) state
    Bit 1 = Pin 2   (Relay 2) state
    Bit 0 = Pin 1   (Relay 1) state

**Any client using the Network Module can Subscribe to
  “NetworkModule/<devicename>/availability” to receive the
  Network module online / offline messages.**

**Any client that wants to receive input or output states changes
  should subscribe to:**
"NetworkModule/<devicename>/input/#"
"NetworkModule/<devicename>/output/#"

**When the Network Module connects to the Broker it will Subscribe
  to the following topics:**
NetworkModule/<devicename>/output/+/set
  The above causes the broker to send the module the "set"
  commands, but won't reflect the modules own "pin" state
  messages back to the module (reduces traffic).
NetworkModule/<devicename>/state-req

**When the Network Module connects to the Broker it will establish
  a "last will" message of "offline" with the will topic**
  “NetworkModule/<devicename>/availability”

**When the Network Module connects to the Broker it will Publish
  to the following topic:**
  NetworkModule/<devicename>/availability Payload: online

nielsonm236, 2020-12-18:
Odd ... when I copied from my notes to the GitHub window it took out "devicename". So everywhere you see "//" above the devicename should appear between the double slash.

yozik04, 2020-12-18:
It is not odd. It strips out HTML. Use backquotes to stop it doing that.

yozik04, 2020-12-18:
> b) With the new definition a client that subscribes to "NetworkModule//input/#" to receive all input changes, and subscribes to "NetworkModule//output/#" to receive all output changes will also see all their "set" commands reflected back to their client (for instance "NetworkModule//output/1/set"). I don't know if this is an issue or not, but stopping this from happening is part fo the reason the current message set was defined the way it is.

This is why you need to subscribe to:
```
NetworkModule/<device>/output/+
```

MQTT has two wildcard symbols: + and #. Copied from documentation:

- A '#' character represents a complete sub-tree of the hierarchy and thus must be the last character in a subscription topic string, such as SENSOR/#. This will match any topic starting with SENSOR/, such as SENSOR/1/TEMP and SENSOR/2/HUMIDITY.
- A '+' character represents a single level of the hierarchy and is used between delimiters. For example, SENSOR/+/TEMP will match SENSOR/1/TEMP and SENSOR/2/TEMP.

#### Any client that wants to receive input or output states changes
should subscribe to (note the `+`):
```
"NetworkModule/<device>/input/+"
"NetworkModule/<device>/output/+"
```

All the rest is right. Thank you.

nielsonm236, 2020-12-19:
I will begin work on incorporating the new MQTT commands tomorrow.

nielsonm236, 2020-12-20:
I've implemented the changes and I'm testing them. In that process I discovered a minor flaw in my performance improvements - I tightened the MQTT loop timing a little too much, causing some MQTT commands to get dropped when a rapid sequence of transmissions are needed (such as when all-on or all-off is commanded). I've fixed that, but unfortunately a lot of performance testing will have to be repeated. I'm coming to the conclusion that this device is well suited to the type of control environments where changes are made in 10's of seconds ... but not in cases where changes are needed several times per second. Not too surprising given the hardware limitations.
I should be able to get a test firmware out to Jevgeni and Carlos in a day or so.

yozik04, 2020-12-20:
Oh wow. I will not be touching anything on the weekends.

nielsonm236, 2020-12-20:
Jevgeni - Can you send an email to me at nielsonm.projects@gmail.com so I can send a test build to you?

nielsonm236, 2020-12-21:
Addressed in Code Revision 20201220 1321. See the Documentation for a complete list of the new MQTT commands.

closed 2020-12-21 by nielsonm236