nielsonm236/NetMod-ServerApp #18: HomeAssistant MQTT Discovery

yozik04, 2020-12-20:
I have spent some time on MQTT Discovery.

Here are required topics and payloads.

I think we need to use mac address as a key here and device name from configuration in human readable fields.

The hardest thing here is buffer size. Will this JSON payload fit? Spaces and line breaks certainly need to be removed.
Device(`dev`) part repeats for all inputs and outputs.

### Switch (Output)
Topic: `homeassistant/switch/<mac>/<##>/config`
Payload:
```
{
  "uniq_id": "<mac>_output_<##>",
  "name": "<device> output <##>",
  "~": "NetworkModule/<device>",
  "avty_t": "~/availability",
  "stat_t": "~/output/<##>",
  "cmd_t": "~/output/<##>/set",
  "dev": {
    "ids": ["NetworkModule", "<mac>"],
    "mdl": "HW-584",
    "mf": "NetworkModule",
    "name": "<device>",
    "sw": "<firmware>"
  }
}
```

### Binary sensor (input)
Topic: `homeassistant/binary_sensor/<mac>/<##>/config`
Payload:
```
{
  "uniq_id": "<mac>_input_<##>",
  "name": "<device> input <##>",
  "~": "NetworkModule/<device>",
  "avty_t": "~/availability",
  "stat_t": "~/input/<##>",
  "dev": {
    "ids": ["NetworkModule", "<mac>"],
    "mdl": "HW-584",
    "mf": "NetworkModule",
    "name": "<device>",
    "sw": "<firmware>"
  }
}
```

-----------------------

So in my case config will look like:
### Switch (Output 01)
Topic: `homeassistant/switch/aabbccddeeff/01/config`
Payload:
```
{
  "uniq_id": "aabbccddeeff_output_01",
  "name": "heating_2floor output 01",
  "~": "NetworkModule/heating_2floor",
  "avty_t": "~/availability",
  "stat_t": "~/output/01",
  "cmd_t": "~/output/01/set",
  "dev": {
    "ids": ["NetworkModule", "aabbccddeeff"],
    "mdl": "HW-584",
    "mf": "NetworkModule",
    "name": "heating_2floor",
    "sw": "<firmware>"
  }
}
```

### Binary sensor (Input 01)
Topic: `homeassistant/binary_sensor/aabbccddeeff/01/config`
Payload:
```
{
  "uniq_id": "aabbccddeeff_input_01",
  "name": "heating_2floor input 01",
  "~": "NetworkModule/heating_2floor",
  "avty_t": "~/availability",
  "stat_t": "~/input/01",
  "dev": {
    "ids": ["NetworkModule", "aabbccddeeff"],
    "mdl": "HW-584",
    "mf": "NetworkModule",
    "name": "heating_2floor",
    "sw": "<firmware>"
  }
}
```

In Home Assistant it will appear this way:
[image]

[image]

nielsonm236, 2020-12-20:
I'm not quite understanding the above, but I will catch up eventually.
The module ID is currently comprised of the devicename with the MAC address appended.
The payloads shown above are Published by the Network Module? The module Publishes the above in response to receiving a "config" Publish?
The MQTT transmit buffer is 200 bytes, and we currently process one command at a time (no queuing in the buffer). The MQTT transmit buffer has to contain the entire MQTT message (but not the link level header, ip header, and tcp header).

caladeira, 2020-12-21:
Hi Jivgeni,

Regarding the fields where you use a name, if I understand correctly, for
each input and output you have to define names (heating_2floor) in the code
before compiling or in fields that you can edit in the web interface, right?
I guess that either way will be too much for the code space available.
Each network module will represent 16 entities, 8 relays and 8 switches.

I know that if we configure a device in the home assistant's
configuration.yaml, you can set the item name there.
I have to read about autodiscovery and find some examples...

Carlos

nielsonm236, 2020-12-21:
In the example above if the intent is that "heating_2floor" is the device Name already entered in the Configuration page I don't think there is a problem with constructing the information in the payload. Of course it implies that an entire module is associated with "heating_2floor" and may have many more inputs/outputs that are needed at that location.

Assuming this is the case, there are only a couple of more fields to add that I think can be hard-coded (in this example "HW-584" and "firmware"). 

The next problem is the size of the payload. Currently the transmit buffer is only 200 bytes, and the example messages here are about 280 to 300 bytes. The transmit buffer could be made a little larger, but probably only about 50 to 75 bytes - but with some effort a little more memory might be made available. Maybe some parts of the payload aren't required?

In the browser transmit code I have the notion of substituting special markers with longer strings as the fields are copied from Flash to the uip_buffer (which is the actual transmission buffer ... up to 440 bytes). So it might be possible to construct something similar to intercept the MQTT message as it is copied from the MQTT transmit buffer to the uip_buffer and insert fields that wouldn't fit in the MQTT transmit buffer. It's a somewhat complicated approach, trading processing time for limited storage resources.

So, those are some thoughts. All this assumes "heating_2floor" is in fact the device Name.

nielsonm236, 2020-12-21:
If I'm on the right track with my comments above the entire "substitute fields during copy" process would be in the mqtt_pal.c file in the mqtt_pal_sendall() function, replacing the current "memcpy(uip_appdata, buf, len);" line with the more complex process. I think it would work ...

yozik04, 2020-12-21:
Sorry it was late night for me and I decided to drop what I have to initiate the discussion.

> In the example above if the intent is that "heating_2floor" is the device Name already entered in the Configuration page I don't think there is a problem with constructing the information in the payload.

It is device name entered on the Configuration page.

> Of course it implies that an entire module is associated with "heating_2floor" and may have many more inputs/outputs that are needed at that location.

Yes. Maybe we need to reserve couple bytes in Config struct to set enabled/disabled state of the inputs and outputs.

> Assuming this is the case, there are only a couple of more fields to add that I think can be hard-coded (in this example "HW-584" and "firmware").

Yes. They should be hardcoded.

> The next problem is the size of the payload. Currently the transmit buffer is only 200 bytes, and the example messages here are about 280 to 300 bytes. The transmit buffer could be made a little [...]

I offered an ideal solution here with shortened key names. There are some key/values that can be dropped or shortened.
`<device>` can be max of 18 chars. It is longer than it is in my example `heating_2floor`
`<mac>` is 12 chars

Ideal payload in compressed version is 316 bytes max + MQTT topic + MQTT header
```
{
  "uniq_id": "aabbccddeeff_output_01",
  "name": "long_device_name_xy output 01",
  "~": "NetworkModule/long_device_name_xy",
  "avty_t": "~/availability",
  "stat_t": "~/output/01",
  "cmd_t": "~/output/01/set",
  "dev": {
    "ids": ["NetworkModule", "aabbccddeeff"],
    "mdl": "HW-584",
    "mf": "NetworkModule",
    "name": "long_device_name_xy",
    "sw": "1.10.1"
  }
}
```

Things we can drop in Device description (`dev`):
* `"mdl": "HW-584",` -15 char
* `"mf": "NetworkModule",` -21 char. If we shorten to `NM` then -11 char.
* `"ids":["NetworkModule","aabbccddeeff"]`. If we shorten to `"ids":"NM_aabbccddeeff"` -15 char.
* `"sw": "1.10.1"` -14 char

Things we can improve in all MQTT:
* If we shorten all MQTT topics. From `NetworkModule/long_device_name_xy/output/01/set` to `NM/long_device_name_xy/o/01/set` and `availability` to `avty` -29 char

It would be good to keep `name` intact as it is used to generate sensor id in home assistant.

To calculate I used https://codebeautify.org/jsonminifier

> In the browser transmit code I have the notion of substituting special markers with longer strings as the fields are copied from Flash to the uip_buffer (which is the actual transmission buffer ... [...]

Too complicated. Let's take simple steps first.

> So, those are some thoughts. All this assumes "heating_2floor" is in fact the device Name.

Your assumption is right.

yozik04, 2020-12-21:
@nielsonm236 
> The module ID is currently comprised of the devicename with the MAC address appended.

yes

> The payloads shown above are Published by the Network Module? The module Publishes the above in response to receiving a "config" Publish?

Yes, published by Network Module for each input and output a separate message is sent. It is not a response. These are published on NetworkModule boot after connection to MQTT, before sending state.

> The MQTT transmit buffer is 200 bytes, and we currently process one command at a time (no queuing in the buffer). The MQTT transmit buffer has to contain the entire MQTT message (but not the link level header, ip header, and tcp header).

Yes, this is my concern. We need to fit in or make the buffer larger. Unfortunately it is not possible to save a lot of space there.

yozik04, 2020-12-21:
@caladeira 
> Regarding the fields where you use a name, if I understand correctly, for
each input and output you have to define names (heating_2floor) in the code
before compiling or in fields that you can edit in the web interface, right?
I guess that either way will be too much for the code space available.
Each network module will represent 16 entities, 8 relays and 8 switches.

Nope. Sensor/switch name will consist of: device name from configuration, entity type (input/output) and entity id(01-08).
No additional space will be needed in eeprom for that.

> I know that if we configure a device in the home assistant's
configuration.yaml, you can set the item name there.

HomeAssistant autodiscovery works the same way. You just pass sensor configuration via MQTT not touching configuration.yaml

yozik04, 2020-12-21:
I have edited the posts above a lot. Please read version on the web as github does not send notifications for edits, if I remember right.

nielsonm236, 2020-12-21:
> Yes, published by Network Module for each input and output a separate message is sent. It is not a response. These are published on NetworkModule boot after connection to MQTT, before sending state.

By "before sending state" do you mean "before sending availability online"?

yozik04, 2020-12-21:
> > Yes, published by Network Module for each input and output a separate message is sent. It is not a response. These are published on NetworkModule boot after connection to MQTT, before sending state.
> 
> By "before sending state" do you mean "before sending availability online"?

Actually can be this way as well:

1. Connects to network
2. Connects to MQTT
3. Subscribes to topics
4. Publishes `availability`: `online`
5. Publishes HomeAssistant discovery configs
6. Publishes state

I do not see any drawbacks in this flow. Question is more. Can device handle commands when it is busy publishing discovery configs and state? If not then availability should be last.

nielsonm236, 2020-12-21:
OK - so by "state" you are referring to pin states. Got it.
Incoming commands can be handled once topic subscription is complete even prior to the discovery configs and publishing of initial pin states. Of course it is all processing time dependent, and it might be possible for a sender to exceed the size of the ENC28J60 receive buffer (6K bytes with queuing) if we are slow to empty it. I don't think this is a problem in actual applications.
I think user experience should be high on our list, even if the code gets hard to read. I'll just continue my practice of "lots of code comments" so we'll know why something was done later on.

yozik04, 2020-12-21:
I recommend you splitting functions that do multiple things into smaller functions. That will keep code cleaner and there will be less demand for comments.

nielsonm236, 2020-12-21:
Yeah, I have to be careful with that. Believe it or not another problem I've battled from time to time in this development is stack overflow. So I've ended up with longer functions to try to avoid creating additional stack depth. And in some cases I've used global variables to limit stack growth. Both are bad coding practices, but necessary in some places.

Also the coding style is not consistent. There are now 4 main code sources in this project, each using a different style. Then add my less-than-professional code (particularly when it comes to proper use of header files), and some of this is quite messy.
The Adam Dunkels code (the uip code) has long functions and a lot of "goto" stuff - but for good reason with regard to small processor speed and stack limitations. The Liam Bindle code (the mqtt code) trends the opposite direction - lots of function calls and a few macros, both of which are less stack efficient.

Once the feature changes calm down a good review of style and efficiency is needed.

yozik04, 2020-12-23:
To save space on the chip I have compacted HTML pages.
#### Configuration page:
https://jsfiddle.net/yozik04/2s8Lm9yh/
#### IO Control page:
https://jsfiddle.net/yozik04/4wspzckh/

To easily drop extra whitespace I recommend putting source code through this tool: https://www.willpeavy.com/tools/minifier/

nielsonm236, 2020-12-30:
This issue addressed in Code Revision 20201230 0411

closed 2020-12-30 by nielsonm236