nielsonm236/NetMod-ServerApp #19: Add timed outputs

nielsonm236, 2020-12-25:
Request is to add a feature that would cause the output to revert to its former state a given number of seconds after a change of state. Perhaps have a byte per output that the user can set to indicate the following action; If output state changes, revert the state after requested number of seconds. This way if the output is normally off, and the user (or MQTT) turns it ON, it will turn back OFF after 1 second (or 2, or 3, etc depending on the setting). Reverse would also be true if the output is normally OFF.

yozik04, 2020-12-28:
We are adding MQTT Auto Discovery in Home Assistant. It is recommended to use Home Assistant for this use case.

nielsonm236, 2020-12-28:
I still want to think about this one a little more. Yes, Home Assistant is a very good alternative. Or Node-Red. But I suspect the request is coming from the perspective of a ''Browser Only" user that doesn't have the means of keeping a server online all the time to manage these devices (and perhaps only has 1 Network Module, in which case adding a server is overkill).

yozik04, 2020-12-28:
you do not have enough EEPROM to hold this info for every output. The only possibility is to add a single duration and a bit map to which outputs this duration needs to be applied.

nielsonm236, 2020-12-28:
If I wanted to use 1 byte per output it would take 16 bytes. We do have that (there are about 46 bytes left which I currently use for debug). Each byte would hold a value from 0 to 256 for seconds. The code would then watch for a change in state, and flip the state back after the number of seconds have passed. So, the user has to set the value to 0, set the resting state (ON or OFF), and thereafter if the user changed the output state the timer would start.
A variation could be to allow 1/2 or 1/4 second resolution by defining the two lowest bits to be partial seconds, but we would have to decide at the start. Another variation would be to allow 1 second resolution, but the upper two bits are minutes.
So there is EEPROM space if we only allow 1 byte per output.
I don't know if this is a good use of the device resources, but two different people have inquired if it would be possible. They didn't describe their use case, so it might be worthwhile asking what their intent is. Frankly if someone wants a pulse output it is an easy circuit to add in hardware.

nielsonm236, 2021-02-28:
Update: I've found that we can store very infrequently changing user inputs in Flash. So if the assumption is made that a user is very unlikely to change "timing" settings very often we could easily store a "per IO" timer value in Flash. 
This could be part of a "Browser Only" build, as this setting is not required for MQTT builds. MQTT implies an upper layer of control, like Home Assistant or similar, that can handle all the complexities of timer based control. 
Referring to the previous comment, and given what I now know about storing user data in Flash, I think a two byte per IO value would be stored (total of 32 bytes). This would allow up to an 18 hour timer with 1 second resolution. Fractional seconds are probably not worth the trouble.

yozik04, 2021-02-28:
0.1 second is needed sometimes.

nielsonm236, 2021-02-28:
I wonder how long such a timer needs to be. I can see someone wanting to have a light on for a few hours (perhaps overnight). If I have 16 bits of timer and want 0.1 seconds of resolution, I have a max time of 1.8 hours. I suppose I could just store a 32 bit value (13 years at 0.1 second resolution!). This is still only 64 bytes. My tests of a Browser Only build (no MQTT) shows I have about 10K of Flash available. So maybe I just do this and move on.

nielsonm236, 2021-03-08:
An update for anyone following this discussion. It looks like we are headed for this implementation: A timer that will revert the state of the Output after a user defined timeout, with timeout units in 10ths of a second, seconds, minutes, or hours.
Some details being worked: 1) Exact GUI presentation, 2) How to bridge a reboot if a timer is running, 3) Can we still maintain the single build approach, or do we need to have parallel builds for "Browser Only" and "MQTT" users.
This may take a couple of weeks, especially if we find we require parallel builds.

nielsonm236, 2021-04-01:
I'm in the final steps of testing this feature for release. I need to add documentation so I'm guessing about a week to go.
Adding this feature required that the firmware get split into separate releases again due to Flash space limitations. One release will the "For MQTT users" and one is "For Browser Only users". The Timers feature will be in the Browser Only firmware. Since MQTT management devices provide plenty of methods to do this kind of timing I think it is appropriate that it be split this way.

nielsonm236, 2021-04-12:
Released with 20210412.1333

closed 2021-04-12 by nielsonm236