nielsonm236/NetMod-ServerApp #21: Elaborating on code size limitation

mkukik, 2021-01-08:
I am wondering... can you elaborate on why 16OUT + MQTT isn't possible?
I did notice you saying it in video and mentioning in doc, but I am wondering if there is much to be done to achieve this.

BTW: beautiful project, I have this board for years, but never got to use it due to its awkwardness... just ordered more of 8OUT.

nielsonm236, 2021-01-09:
16out without MQTT works fine. The problem is (or was) that the code size grows too large for the 16out version plus MQTT.
 
I've been working with another developer who is very talented with javascript. He has reduced the size of the html pages significantly, and it now looks like we will be able to release a version that allows you to specify each pin as input or output, all the way up to 16 pins. We are still working on that, but should finish up in a week or two.

By the way, you can buy the 8 pin version, and when you load this code it will work on all 16 pins. The hardware is identical for the 8 and 16 pin versions, and only the factory firmware limited it to 8 pins.

mkukik, 2021-01-09:
Yeah, I've seen your note somewhere stating that 8 pin is actually 16 pin version limited. Hence why I rushed to buy 3x 8pin modules :)

I've been looking through the code and I came to conclusion that you should consider this:
Have just rest api + MQTT on the module and create sample single page external html app that will interact with the rest api.

It would be sufficient for debugging purposes or small standalone projects (not as a part of some automation).
Usually when you integrate it into some automation you will rarely even use UI.

This could significantly simplify and reduce the code.
I could even help with the single page HTML app.

mkukik, 2021-01-09:
Also I understand why you have opted for same pattern of URL to control Relay On/Off as in original implementation, but I think if there is just rest api it could be made more transparent.

Old:
//IP:port/00 = Relay 1 OFF
//IP:port/01 = Relay 1 ON

New:
//IP:port/R01/0 = Relay 1 OFF
//IP:port/R01/1 = Relay 1 ON
or whatever suits better...

nielsonm236, 2021-01-18:
I think I need to keep the "old" method as I suspect some people have developed apps that use it. I don't know this for sure, but on line research suggests it.
So the question becomes "Can both an old and new be supported at the same time". I have to say No for now - as we are painfully low on Flash memory now that we've combined MQTT and Browser operations all into one code load. To do these sorts of things may push me back to having two separate code threads ... one for Browser/Misc Stuff and and one for Browser/MQTT.
So I'll keep this open, or open a new issue and we'll see where it goes.

nielsonm236, 2021-01-23:
The original issue of providing 16 outputs with MQTT has been addressed with release 20210123 1257. The additional commentary about changing the REST interface cannot be addressed at this time due to memory limitations. While the new release combines all functionality into one firmware load, I may consider taking a step back to a "Browser Only" version (with no MQTT) that can address requests like alternative REST implementations ... my assumption being that if REST is being used no MQTT interface is needed. I'll open a new Issue with the above commentary.

closed 2021-01-23 by nielsonm236