nielsonm236/NetMod-ServerApp #22: 16 inputs are not fully functional

Sky-Jumper, 2021-01-17:
Good day! Thank you so much for your work, he helped me a lot with this board!
There is a problem with the firmware for 16 inputs, inputs 1-8 do not work, immediately after the firmware they are marked in red, no settings could make them work. On the same board, the firmware for 16 outputs works fine, tell me what to do?

Добрый день! Огромное спасибо за ваш труд, он очень помог мне с данной платой!
Есть проблема с прошивкой на 16 входов, не работают входа 1-8, сразу после прошивки отмечены красным, никакими настройками не получилось заставить их работать. На этой же плате прошивка на 16 выходов отлично работает, подскажите что сделать?

nielsonm236, 2021-01-18:
This is a bug. But we are close to releasing the new user interface that will allow each pin to be individually set as input or output, so we'll make sure it works in the new release.

nielsonm236, 2021-01-23:
This is addressed in the 20210123 1257 release. Note that the new GUI allows every pin to be individually assigned as an Input or Output.

closed 2021-01-23 by nielsonm236