nielsonm236/NetMod-ServerApp #23: Button in config-page(with option to save) for inverted gpio outputs

balou1974, 2021-01-20:
Hello,

firstly thanks for your great work! Your work makes the hw-584 card much more usable !

I am using a sixteen channel relay-card (should be saint smart with optocouplers).

My problem is that the gpio outputs need to be inverted have the right result that is shown on the web-interface
(mqtt commands and weblinks are wrong, too).
It would be great to have a button on the config-page to set this permanently to inverted or not inverted outputs.

I would have fixed this myself but I had trouble using the stvd to import the project to a workspace ....

Thanks a lot for your efforts!

nielsonm236, 2021-01-20:
Which version are you running? If you can send a screen shot of the Configuration page I can tell right away. If you have a version with the 6 character Config field it provides the ability to set Output Invert (see the documentation or Help screen). If you have an earlier version I suggest you upgrade to get the functionality.
AND ---> Soon I will release a new version that lets you set each pin as an input or output, and lets you set the Invert function for each pin. A couple of co-developers worked up a much more flexible user interface, and I worked up the code behind it. I'm hoping we get it released before the end of January.

balou1974, 2021-01-20:
Shame on me, I shoud have read the documentation! Please close the issue! I found the Config Part under Configuration! Simple solution: change Config to 102000.

Everything is working fine now, except the relay numbering but that is not a problem ;)

closed 2021-01-20 by balou1974