nielsonm236/NetMod-ServerApp #25: Alternative REST commands

nielsonm236, 2021-01-23:
mkukik: "... I understand why you have opted for same pattern of URL to control Relay On/Off as in original implementation, but I think if there is just rest api it could be made more transparent.
Old:
//IP:port/00 = Relay 1 OFF
//IP:port/01 = Relay 1 ON
New:
//IP:port/R01/0 = Relay 1 OFF
//IP:port/R01/1 = Relay 1 ON
or whatever suits better..."

nielsonm236, 2021-01-23:
At this time there isn't enough memory to add a parallel "more RESTful" interface. But in combination with Issue #26 it might be possible.

yozik04, 2021-01-26:
OR:
```
//IP:port/on/65535 = All Relays ON
//IP:port/off/65535 = All Relays OFF

//IP:port/on/8 = Relay 4 ON
//IP:port/off/8 = Relay 4 OFF

//IP:port/on/16 = Relay 5 ON
//IP:port/off/16 = Relay 5 OFF
```

65535 - is a bit mask which relays to trigger to the requested state.

nielsonm236, 2021-02-28:
Update: I will look at this as part of a "Browser Only" build. There isn't much use for REST commands in a MQTT environment (other than they are useful for test ... but the existing command set can be used for that).

mkukik: Is there a reason you suggested the format above? Is it commonly used? Or used in an application you have? Are you using MQTT?

yozik04: It is easier and slightly less code if the values (like 65535, 8, 16) are all the same field width. Not a deal breaker either way, but a few bytes of code difference. Of course, if this is a Browser Only build a lot of Flash is available so my argument is weak.

mkukik, 2021-03-01:
While bit mask is nice from programmer's perspective it is not intuitive in my opinion... Also if we have fixed number of IOs it is unnecessary to deal with 65535 number...

No, the format I suggested is not based on any standard out there (as far as I know) it is only my gut feeling (working with APIs over the years).

Also, the format may not make sense if we consider that IOs can be either input, output or sensor.

In my option the first in the url should be "what to change" and second "what to do". 0 & 1 is self explanatory and can be easily extended with 3, 4, 5, ... if we need to control (or get) other than ON or OFF status (delayed On, delayed Off maybe?).

Since IOs can now be changed to either input,  output or sensor the format of urls could change accordingly:

//IP:port/I01/0 = Read

//IP:port/O01/0 = Relay 1 OFF
//IP:port/O01/1 = Relay 1 ON
...
//IP:port/O99/0 = All Relays (only) OFF
//IP:port/O99/1 = All Relays (only) ON

//IP:port/T01/0 = Temperature read

I am not currently working with this module (no time) and only following github discussion so I don't have practical insight into the inner workings... so my opinion is solely a hypothesis and I sincerely hope that you will use your best judgement. I am merely just giving my 2 cents.