nielsonm236/NetMod-ServerApp #27: "HA Auto" adn "MQTT" flags

Sky-Jumper, 2021-01-24:
Hello!
If you set the "HA Auto" flag, then when you save the configuration, the "MQTT" flag becomes set, and while "HA Auto" is enabled, "MQTT" will not be disabled in any way.
Is this the way it should be, or is it a mistake?

nielsonm236, 2021-01-24:
That's the way it works. HA Auto (Home Assistant Auto Discovery) can only work over MQTT, so setting HA Auto requires that MQTT be set. Per the manual Page 13: HA Auto: The Home Assistant Auto Discovery setting enables the Network Module to send MQTT Auto Discovery Publish messages to your Home Assistant server. Checking this setting will automatically enable MQTT (and the MQTT checkbox will automatically be set). Do not enable this setting unless you are operating in an MQTT environment with Home Assistant.

nielsonm236, 2021-01-24:
Also note that you can have MQTT enabled with HA Disabled. That will be a case where:
a) Home Assistant is not used - some other MQTT Client is used for MQTT control of the Network Module (Node-Red comes to mind).
b) Home Assistant IS used, but its Auto Discovery function is not used. This is fairly common where a Home Assistant user wants to manually set up and remove connections.

nielsonm236, 2021-01-24:
I appreciate your testing this. If you find anything at all please ask. Even if it is already in the manual the questions can help point me to areas that could use better documentation.

Sky-Jumper, 2021-01-24:
Thanks for the explanation, I assumed something like that.
Your firmware is very cool, it covers all my needs, even more.

closed 2021-01-24 by Sky-Jumper