nielsonm236/NetMod-ServerApp #28: Problem due to empty MQTT Server

Sky-Jumper, 2021-01-25:
Such a problem:
There are two boards, one is configured for 16 outputs with inverted outputs, the second for 16 inputs, in the settings I completely cleared the MQTT Server address and login / password on both boards and saved the settings.
It turned out like this:
The board configured as 16 inputs is no longer on the network, the IP scanner does not find it, after resetting to the factory settings, it worked, set the IP address 192.168.0.250, saved everything to the input, saved it, the board again does not find it, it turned out in this order - set everything to the input, saved, and only then configured the IP address;
Board configured as 16 outputs: in the "Name" field it was "Relay_16" became "0 (Windows NT 10.0;"; in the "MQTT Server" field before clearing the field it was "192.168.2.250" became "192.168.2.88"; the "MQTT" field Username "should have become empty, but the old value remained; the" MQTT Password "field should have become empty, but it has" age = 0Upg "

nielsonm236, 2021-01-26:
Let me try this here. I don't think I tested "empty MQTT server" ... I always had at least 0.0.0.0

nielsonm236, 2021-01-26:
In my test leaving the MQTT IP Address blank caused the module to become unresponsive. This is definitely a bug. For now, at least keep 0.0.0.0 in the MQTT IP Address field until this is fixed.

With regard to the field corruptions I am not sure what happened, but I will bet the empty MQTT IP Address caused the fields returned by the Browser to be parsed incorrectly. To get back to a working module look in the Manual section "Alternative Way to Force Defaults or Downgrade Firmware". IGNORE THE DOWNGRADE FIRMWARE PART - just do the clear EEPROM part by erasing the EEPROM revision and Magic numbers. The firmware will boot as if this is a new module and should come up with defaults. But again - for now put 0.0.0.0 in the MQTT IP Address.

nielsonm236, 2021-01-26:
Adding a little more test information to help find the bug:
- A blank IP Address followed by Save resulted in a reboot, and restoration of the previous IP address
- A blank Gateway Address followed by Save DID NOT result in a reboot. When refresh was clicked the Module device name was corrupted, but the Gateway Address was restored to its previous value.
- A blank Netmask followed by Save resulted in a reboot and restoration of the previous Netmask.
- A blank Port number followed by Save resulted in a reboot, and restoration of the previous Port number.
- A blank MAC followed by Save resulted in a reboot, and restoration of the previous MAC value.
- A blank MQTT IP Address followed by Save DID NOT result in a reboot. Refresh had no effect ... as if the module or javascript were in a loop. But the Independent Watchdog did not fire, so the main loop must still be running. A power cycle brought the module back with its previous MQTT IP Address intact.
- A blank MQTT Port number followed by Save DID NOT result in a reboot. When refresh was clicked the MQTT Port number  was restored to its previous value.
- Blank MQTT username and password work OK.
In each of these cases we should have seen a "Please match the requested format" message rather than a reboot or erroneous result. I suppose a reboot and restoration of previous content is possibly OK, but might be confusing if someone made several entries, some were accepted, and other remained "old" without any messaging. Need to examine this.

nielsonm236, 2021-01-26:
Sky-Jumper - I found and fixed two bugs that change the behavior. I updated the code on GitHub. These fixes do not affect the missing "Please match the requested format" messages but might address the field corruption you saw.

Sky-Jumper, 2021-01-26:
> With regard to the field corruptions I am not sure what happened, but I will bet the empty MQTT IP Address caused the fields returned by the Browser to be parsed incorrectly. To get back to a working module look in the Manual section "Alternative Way to Force Defaults or Downgrade Firmware". IGNORE THE DOWNGRADE FIRMWARE PART - just do the clear EEPROM part by filling the EEPROM revision and Magic numbers with 0's. The firmware will boot as if this is a new module and should come up with defaults. But again - for now put 0.0.0.0 in the MQTT IP Address.

No need to reprogram, it was enough to reset to default settings (by holding the button on the board for 10 seconds)

Sky-Jumper, 2021-01-26:
> Adding a little more test information to help find the bug:
> 
Now I cannot test the module in different modes, I will describe in more detail what manipulations were performed:

The network card is configured for the 2nd subnet

Module configured for 16 outputs:

The initial settings are as follows:
Name - Relay_16
IP Address - 192.168.2.250
Gateway - 192.168.2.1
Netmask - 255.255.255.0
Port - 8080
MAC Address - (by default, the last digits are "00")
Full Duplex - not checked
HA Auto - checked
MQTT - Checked
MQTT Serwer - 192.168.2.88
MQTT Port - 1883
MQTT Username - admin
MQTT Password - 123

All IOs set as inverted output

Changed the settings to the following:
Name - Relay_16
IP Address - 192.168.0.250
Gateway - 192.168.0.1
Netmask - 255.255.255.0
Port - 8080
MAC Address - (by default, the last digits are "00")
Full Duplex - not checked
HA Auto - checked
MQTT - Checked
MQTT Serwer - {Empty}
MQTT Port - 1883
MQTT Username - {Empty}
MQTT Password - {Empty}

All IOs set as inverted output

then clicked "Save"

The inscription "Wait 5 sec" did not appear, after reconfiguring the network card to the 0th subnet and opening 192.168.0.250:8080 I saw this:

Name - 0 (Windows NT 10.0;
IP Address - 192.168.0.250
Gateway - 192.168.0.1
Netmask - 255.255.255.0
Port - 8080
MAC Address - (by default, the last digits are "00")
Full Duplex - not checked
HA Auto - checked
MQTT - Checked
MQTT Serwer - 192.168.2.88
MQTT Port - 1883
MQTT Username - admin
MQTT Password - age = 0Upg

All IOs set as inverted output

16-input module:

The initial settings are as follows:
Name - 16_inputs
IP Address - 192.168.2.251
Gateway - 192.168.2.1
Netmask - 255.255.255.0
Port - 8080
MAC Address - (by default, the last digits are "01")
Full Duplex - not checked
HA Auto - checked
MQTT - Checked
MQTT Serwer - 192.168.2.88
MQTT Port - 1883
MQTT Username - admin
MQTT Password - 123

All IOs are set as input, no inversion

Changed the settings to the following:
Name - 16_inputs
IP Address - 192.168.0.251
Gateway - 192.168.0.1
Netmask - 255.255.255.0
Port - 8080
MAC Address - (by default, the last digits are "01")
Full Duplex - not checked
HA Auto - checked
MQTT - Checked
MQTT Serwer - {Empty}
MQTT Port - 1883
MQTT Username - {Empty}
MQTT Password - {Empty}

All IOs are set as input without inversion

then clicked "Save"

The inscription "Wait 5 sec" did not appear, the module is not visible.

In the evening I checked from another computer: the first module shows corrupted settings, the second module is not visible. I did not remove the power supply, reconfigure the first module, reset the second to the default settings and reconfigure.

P.S. Is it possible to add the contents of the "Name" field to the <title> tag so that the browser tabs can see which module is open (for example, instead of "IO Control" write "{Name} IO Control")

nielsonm236, 2021-01-26:
I will add your last comment about "(Name) IO Control" as a new issue. For now please make sure you are using the code load that includes fixes from last night. It will report as revision  Code Revision 20210126 0527. AND you must make sure the MQTT IP Address field is not blank. Let me know if your test results above were with Code Revision 20210126 0527.

nielsonm236, 2021-01-26:
I think if you make sure you are on the 20210126 0527 revision and that the MQTT IP Address has 0.0.0.0 in it you will not see the problems described above. And, of course make sure the two modules have different MAC addresses.

Sky-Jumper, 2021-01-26:
>  Let me know if your test results above were with Code Revision 20210126 0527.

Test results above were with Code Rev 20210123 1257

nielsonm236, 2021-01-26:
OK - Switch to the more recent code and you'll get better results. We will fix the "blank field" issue shortly. MQTT Username and Password can be blank, but all other fields cannot be blank for now.

nielsonm236, 2021-01-26:
I'm integrating and testing a fix for the blank fields now. Hope to have results within 24 hours.

nielsonm236, 2021-01-27:
Fixed in Code Revision 20210127 1112

closed 2021-01-27 by nielsonm236