nielsonm236/NetMod-ServerApp #31: "Please enter requested format" message does not scroll with browser page in Firefox

nielsonm236, 2021-01-26:
Try this: Using Firefox delete the last character of your MAC address, then click Save. A "Please enter requested format" message appears. Now scroll the page up an down. The "Please enter requested format" popup does not stay attached to the MAC field.

If you do this same thing in Chrome the "Please enter requested format" popup stays attached to the MAC field.

I didn't try other browsers.

yozik04, 2021-01-26:
This is the browser bug

nielsonm236, 2021-01-26:
There are two browser bugs. Issue #28 (blank fields hose things up), and #31 (this one).

yozik04, 2021-01-26:
I mean I can not fix it in javascript or HTML.

closed 2021-01-26 by yozik04

nielsonm236, 2021-01-26:
OK - Thanks. This is not critical, and only seems to appear in Firefox.