nielsonm236/NetMod-ServerApp #33: Request that the browser tab include the Device Name

nielsonm236, 2021-01-26:
From Sky-Jumper: Is it possible to add the contents of the "Name" field to the <title> tag so that the browser tabs can see which module is open (for example, instead of "IO Control" write "{Name} IO Control")

nielsonm236, 2021-01-26:
I will look at this ... but the IOControl page has the Device Name as the very first field at the top of the page. I'm assuming this request is so that a browser with multiple tabs open, each one to a different Network Module, will be easier to navigate just by looking at the tabs (and not the page itself). Is this correct?

yozik04, 2021-01-26:
He wants <title> to include device name.

nielsonm236, 2021-02-28:
Adding this to the next build. Need to test, but looks like it only adds 14 bytes to the code :-)

nielsonm236, 2021-02-28:
During test I found and fixed a bug. So the total impact on code is 20 bytes. I decided to make the title look like this:
DeviceName: IO Control
I also added it to the Configuration page
DeviceName: Configuration
So I will close this issue and you can expect this when the next release is done. This was a nice idea for users that have multiple devices and use Browsers for control. Not needed for MQTT users, but it doesn't impact code much so it will appear in those builds too.

closed 2021-02-28 by nielsonm236

nielsonm236, 2021-03-08:
Update for anyone following this issue: I have this implemented in the branch that is being used for the IO Names and Timers changes. For that reason the release of this change may be delayed a couple of weeks - perhaps to the end of March 2021.