nielsonm236/NetMod-ServerApp #49: Real Time Clock and Time-of-Day Output Changes

nielsonm236, 2021-03-09:
This request from YouTube subscriber Miroslaw Walasik.
The request is to have some form of Real Time Clock and the ability to turn outputs on and off based on Time Of Day when used only with a Browser (ie, no MQTT or upper level timing controllers).
There are a couple of ways to do this: 1) Add an RTC hardware device, 2) Use NTP (which requires internet access). And, of course, it requires the user interface to allow the user to make time settings on a per output basis.
In the spirit of "very low cost relay controller" NTP seems the best way to do this, although I am sure there will be some negative feedback about letting the device access the internet as it has no security features.
I am also concerned that this is somewhat beyond the spirit of this device just due to the amount of user interface required. The GUI is one of the most memory "expensive" aspects of the GUI, so we need to be cautious about this kind of addition.
Still, it is a nice idea if it can be implemented so this issue is opened for future consideration.

mkukik, 2021-03-09:
I down vote this feature request... in my opinion it is waste of your time and resources... since this is LAN controller it's sole purpose is to be controlled over LAN and not to be programmable over LAN to a crazy extent...
We could endlessly request to extend it to be controlled by ambient sensor, sound sensor and whatnot. In my opinion there are devices more suited for such purpose.
If you still intent to do this or else I would keep it out of main build at least... just my 2 cents (gosh I feel I am being very negative... sorry)

mkukik, 2021-03-09:
maybe better way would be to have ability to program IOs so one "I" can turn on/off several "Os"... this way external device (arduino) could control that pin based on several sensors and some logic?

nielsonm236, 2021-03-09:
Thank you for the input. We've really stretched the ability of the device already, and I agree its real purpose is just an IO interface ... not a replacement for lots of logical functions. So this one will have to stay in the "think about it" category for awhile. I also think that once someone has an Arduino connected to their network they already have the ability to send commands to the Network Module and do just about any timing related thing they want to do ... assuming they know how to write that code. I need to be more resistant to adding functions that need to be provided from another device.  :-)

nielsonm236, 2021-03-09:
OK - I understand now. Also please look at my comment on Issue #26. I would like to explore your offboard GUI idea.

yozik04, 2021-03-09:
I think if Arduino needs to talk to this module it should talk via serial or I2C. Not via some Input -> *Output mappings.

yozik04, 2021-03-09:
TCPSerial actually can reuse same communication logic that Arduino would use.

nielsonm236, 2021-03-09:
The problem is that those interfaces (serial, I2C) are not available on Network Module. They do not come out to pins. Only the 16 IO pins and the SWIM interface come out to accessible pins. So, where I've implemented the DS18B20 Temperature Sensors I actually had to use IO 16 and write the Maxim single wire interface to run on that pin (when the user enables that function). I had in mind using two more pins for an I2C interface, but I'm becoming convinced that is not a good idea ... not because of the pins, but because it is probably not aligned with the intended device functionality.

nielsonm236, 2021-03-09:
Ooops - I just realized that was yozik04 with that comment ... I think you already knew the above info.

nielsonm236, 2021-03-09:
By the time you do all this unique interface stuff you could as easily implement HA on a pi and have a very capable management system running on your local network.

mkukik, 2021-03-09:
> By the time you do all this unique interface stuff you could as easily implement HA on a pi and have a very capable management system running on your local network.

I agree... hence why my first comment... however my second thought was to make initial idea and beyond possible through additional micro controller (MCU) without dependency on network connectivity and central HUB (initial idea from Miroslaw Walasik)

yozik04, 2021-03-10:
Too much hassle to have a separate microcontroller. Even if you hook a separate micro to our module you have no way to sync RTC switch winter/summer times. My experience with smart house showed that "run something at this time of a day" is not a good way to do things. Days are different. Some are dark, some are light. On weekends you want a different schedule. All this complicates things too much.

nielsonm236, 2021-03-10:
And ... I'm trying to imagine what the GUI looks like to manage the many variations on "1 input pin controls 1 or more output pins".