nielsonm236/NetMod-ServerApp #51: file missing

megamarco833, 2021-04-01:
hi, thanks for the project first of all!
i used the first release, and now i want to update with latest one.
but with latest version i'm not able to find:
NetworkModule.stp - The STVP project file
NetworkModule.sx - The NetworkModule binary file

i had in previous version but i do not see anymore inside project folder, i'm wrong?

thanks!

second question, do you think that it's possible to implement a way that after first flash with your software, it will be possible to update the software without flashing again with USB programmer ST-Link V2 but using web page?
thanks again

nielsonm236, 2021-04-01:
Hi - I'm using the GitHub "Release" method now for distributing the files. You'll find the latest .stp and .sx in the "Releases" area of the site. Look on the right hand columns. You should find the files there.
With regard to flashing via ethernet ... unfortunately because the hardware design didn't using the actual SPI hardware of the device and instead used regular IO pins for a bit-bang interface to the ENC20J60 there isn't a way to use the boot loader on the device. So for now I'm not seeing a way to do this. I do continue to think about it as I would also like that capability.
Mike

nielsonm236, 2021-04-06:
megamarco833 - Did this work for you?

megamarco833, 2021-04-07:
hey ciao @nielsonm236 
first of all sorry for my late answer!
yes, after your post i see that you used GitHub "Release" method and I see the:
NetworkModule.stp - The STVP project file
NetworkModule.sx - The NetworkModule binary file

so it was my guilty that i didn't see them.
In the next days i will try to upgrade my firmware that was already flashed with your old version.
that's a pity that every time that i want to upgrade i need to use the usb programmer, but you explained the limitation of the hardware...
thanks again and sorry for late answer!

nielsonm236, 2021-04-07:
OK - I will close this issue, but please let me know if there are further problems. I have run into the same problem reprogramming once a module is installed somewhere ... it can be a pain to upgrade the firmware later on. In one case I left a long 4 pin ribbon attached so that I could get my laptop close enough to reprogram the module without having to physically remove the module. But that won't work in other situations. Perhaps someday I can come up with a method of remote programming.

closed 2021-04-07 by nielsonm236