nielsonm236/NetMod-ServerApp #59: Integration Web Relay Con V1

sg2710, 2021-04-19:
Hi nielsonm, i found this https://github.com/dpiegdon/SainSmart-REY-NET-v1.1.
He fixed the MAC Bug also. 
Is it possible to merge the your Code that is working with V1 also?

nielsonm236, 2021-04-19:
It doesn't look like he has worked on it in 4 years ... and I'm pretty sure the v1.0 isn't manufactured anymore (I don't have one and early on this project last year I was unable to get one). So, the question "Is it possible?" Anything is possible with enough effort.  But I don't think I personally want to take this on. There is enough difference in the architecture that I believe it would take 6 months or more. HOWEVER - My code is free for the taking if you want to tackle the project.
Mike

sg2710, 2021-04-19:
Hi Mike, thank you for your reply.
BR Stefan

closed 2021-05-09 by nielsonm236