nielsonm236/NetMod-ServerApp #60: MQTT status fail and  home assistant issues

MarcBex, 2021-04-19:
Hello Nielson,
on verson 0.118 of HA i had programmed output with success but i was not able to read any inputs 
The folowing code configuration.yaml works well:
switch:
  platform: command_line
  switches:
 ######################################################
# IP 192.168.105.105 config

server_off_outlet_01:
command_off: curl http://192.168.105.105:8080/00
friendly_name: 'Server Off 105 n°: 01'
server_on_outlet_02:
command_on: curl http://192.168.105.105:8080/01
friendly_name: 'Server  On 105  n°: 01'

I am now on the last version and i would like use MQTT and HA auto discover.
Unfortunatly i am not good enought to progam.

Can you please show me (and for most of begginers) how to do with real code and the place they have to be.
for examples the configuration.yaml code if needed, with the new HA configuration,
how to read HW-584 datas and temératures on binary sensor or switches.

On my MQTT status only 3 are green and i have no idea how tu setup them.

PS: HW-584 flashing code instructions was clear and details, so that was easy to do.
If needed i can let my email if it is allowed on this site

Regards

nielsonm236, 2021-04-19:
I'm very new to Home Assistant and MQTT myself ... so I will try to help but a beginner myself. Hopefully another user can chime in as well. I had the advantage of advice from fellow developers that are very experienced with it.

First, let's make sure you get green indicators on the Configuration menu. Nothing works until that happens. The indicators are automatic "status" lights. But it can take 30 seconds or more for the MQTT connection to fully establish, and it may restart once or twice during power on. So, refresh the Configuration menu a couple of times during the first minute and you should see those indicator turn green if your Broker is operating. Let me know how that works for you. And, let me know what Broker you are using. I'm using Mosquitto.

nielsonm236, 2021-04-20:
I'm creating some screen shots to show how I set up a Network Module in Home Assistant. It is all HA GUI based (not YAML ... although if you are familiar with YAML this may create the initial files for you to edit). Write to me at nielsonm.projects@gmail.com and I'll send what I have. I need another hour or so to finish it.

nielsonm236, 2021-05-09:
Marc's issue was eventually resolved. This was a rather confusing process (at least confusing for me) of helping Marc to get Home Assistant and the HA Mosquitto Add-in installed and working properly on his host. Many thanks to Carlos for helping Marc with this. No real issues with the Network Module, although as his first target it was reasonable to assume there might be a problem. I will close this issue and sort through the emails to see if any sort of HA install guide can be provided ... although that is not the purpose of this project.

closed 2021-05-09 by nielsonm236