nielsonm236/NetMod-ServerApp #61: dht11 temperature and humidity sensor

atilim07, 2021-04-21:
hello if I ask
Can you integrate dht11 temperature and humidity sensor module?

nielsonm236, 2021-04-21:
I've looked at the DHT11 spec and the short answer is "Yes".
The longer answer:
- In the MQTT version of the software we are, once again, very low on memory. So it might not fit unless we only provide the option of "DS18B20, or DHT11". But I will have to analyze further.
- In the Browser Only version there is plenty of memory, so I know it will fit there.
- I'm trying to finish up a couple of other things, so I will come back to this in a week or two.

In the meantime:
- From the spec it looks like the device cannot be operated in parallel with other DHT11s. Is this your understanding also?
- If the above is true my intention would be to implement a single device on a single pin. Do you think this is adequate? Or would the typical application require more than one DHT11?

atilim07, 2021-04-21:
i plan to run dht11 on singel pin thanks in advance for the improvements

nielsonm236, 2021-05-08:
Another user has requested the DHT22. That is a more precise version of the DHT11, and uses the same IO interface. So, if either is implemented both will work. I have ordered DHt11 devices to experiment with. Still don't know if it will fit in code.

nielsonm236, 2021-05-12:
Even though I've ordered DHT11 devices for some experiments with this feature I'm starting to have doubts that this is a good way to go. The DHT11/22 devices can only be attached on dedicated pins ... no means of attaching more than 1 without sacrificing more pins. I found an alternative device (based on the AHT10) that connects with I2C. If I2C connectivity is used that then allows the connection of multiple sensors and other devices without using more IO pins. For those that requested this feature would this be of interest? This link is an Amazon offering of the device.
https://www.amazon.com/AITRIP-Precision-Temperature-Measurement-Communication/dp/B08SGBHJHD/ref=sr_1_8?dchild=1&gclid=CjwKCAjw1uiEBhBzEiwAO9B_HS3SyPkAhlpw4TcSDGC9bG-r4xKBBelefrxeBWiP-6PIsrfGvnCd_xoC9R8QAvD_BwE&hvadid=238245127404&hvdev=c&hvlocphy=1014452&hvnetw=g&hvqmt=e&hvrand=3353284741077870345&hvtargid=kwd-316322552610&hydadcr=18063_9813246&keywords=i2c+temperature+humidity+sensor&qid=1620770365&sr=8-8

caladeira, 2021-05-12:
Hi to all,

I might have another suggestion: BME280
This sensor provides temperature, humidity and pressure.
It uses I2C but can also use SPI interface.

https://pt.aliexpress.com/item/1005002422171343.html?spm=a2g0o.productlist.0.0.38842a39gObY2V&algo_pvid=05bc92da-63b4-4a77-8e5d-32cd21679d25&algo_expid=05bc92da-63b4-4a77-8e5d-32cd21679d25-0&btsid=0bb0624416208259044007443e4819&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

nielsonm236, 2021-05-12:
Wow - now that's interesting. Very inexpensive and much more accurate.

nielsonm236, 2021-05-14:
Jose Vieira discovered some information that shows we've got to be careful about these sensors. At a minimum we want a temperature + humidity sensor. Having barometric pressure would be a plus. But:
BME280 measures temperature, humidity, and barometric pressure
BMP280 measures temperature and barometric pressure (but not humidity)
AND - many vendors don't know what they are selling and mid-identify the product.
In general the BME sensor is more expensive.
See these links:
https://goughlui.com/2018/08/05/note-bosch-sensortec-bmp280-vs-bme280-sensor-confusion/
https://www.meteocercal.info/forum/Thread-Bosch-Sensortec-BMP280-vs-BME280-Sensor-Confusion
https://imgur.com/jnMkxKy

caladeira, 2021-05-14:
I have a few bme280 and bmp180 (older version).

Nothing better than the devices datssheet:

https://cdn-shop.adafruit.com/datasheets/BST-BMP280-DS001-11.pdf
https://cdn-shop.adafruit.com/product-files/2652/2652.pdf

sjorsBe, 2022-10-07:
Is there room for support of an PCF8574? It would make the use of extra pins configurable…

kr 

sjors

nielsonm236, 2022-10-08:
@sjorsBe Please open the PCF8574 question as a new Issue. I'll answer there.
Thanks
Mike