nielsonm236/NetMod-ServerApp #64: DHCP support

colin-purcell, 2021-04-29:
Do you have any plans to add support for DHCP?

I have a strong preference toward DHCP reservations on my networks (instead of static IPs), and enjoy the following benefits:
-moving devices quickly between networks for safe troubleshooting
-easier "configure and move elsewhere" situations (as my network is isolated/locked-down for my IoT devices)
-centralized GUI for quick IP view/management
-simpler major network/subnet changes
-remote management via a single CLI for scripting fun

For some users, this feature might also benefit from local discovery (with NETBIOS or Zeroconf) but that is outside the scope of this request, and not something I personally need.

Thank you all the great work!

yozik04, 2021-04-29:
DHCP is too heavy. It will not fit.

nielsonm236, 2021-04-29:
The DHCP client code itself is a fairly simple "code concept", although I agree even "simple" likely won't fit is the few hundred bytes remaining, particularly in the MQTT version of the code.
Colin - A couple of questions:
- When devices join your network, how do you determine their IP address? Via an "arp -a" in a command window? I'm just curious about your method of determining how to access an added device.
- Do you expect name service for small devices like this? Or is using the IP address adequate? The task and concerns expand significantly if any DNS aspects come into play.

colin-purcell, 2021-04-29:
> ...even "simple" likely won't fit is the few hundred bytes remaining, particularly in the MQTT version of the code.

Ah, I hadn't even thought of size constraints, primarily because DHCP was available as a default on other embedded projects I've used (e.g. [espurna](https://github.com/xoseperez/espurna)), but that just goes to show my ignorance in the problem space. With my setup, MQTT was a "must".
> * When devices join your network, how do you determine their IP address? Via an "arp -a" in a command window?

I pop open my router's web UI (pfsense), and look for the most recent DHCP lease. Of course your arp suggestion would work fine too, but rather than verifying the MAC address I find checking timestamps to be simpler (despite a couple more "clicks"). Even with dozens of devices listed, there's enough information displayed that a quick glance reveals just a single likely IP (though in a very cluttered network, and very rare cases, coincidence might provide two to try).
> * Do you expect name service for small devices like this? Or is using the IP address adequate?

For my own setup, my router can also perform DDNS updates on behalf of clients with DHCP reservations, so DHCP alone is sufficient. The few devices I perform DDNS updates from directly are the ones that connect to my VPN permanently or periodically from outside my home network (laptops, family members' routers, or other "fat" devices I manage remotely). The ISC DHCP server also supports DNS updates on behalf of clients, if someone wanted to replicate what I'm doing but didn't want to replace their router.

nielsonm236, 2021-04-30:
OK - I think size of the code is still a problem, but I will keep this on the list until I can do more investigation.