nielsonm236/NetMod-ServerApp #65: Temperature sensor´s odd behaviour at negative temperatures

jmcvieira1, 2021-04-30:
Hello i´m using four sensor´s DS18B20 to monitor temperatures on a two door refrigerator, and came across with
some spikes in the graphic on domoticz but only on negative temperatures, i will try to explain when reading negative temperatures for example
-21,8 next -21,9 next it should red -22 but it jump´s to -21 only to jumping again on the next reeding to -22,1, first i thought it was a sensor problem with the low temperatures, but i change the sensor and the behaviour is still the same, this symptom only occurs in temperatures bellow 0º Celsius, i´m using Code Revision 20210413 1254 with nodered to translate the mqtt messages to domoticz.
Does anyone came across with something like this?
[image]
[image]
[image]

nielsonm236, 2021-05-01:
I will check this tomorrow. I just fixed a couple of errors in my math and it may have addressed this. I will test these temperatures specifically to see it is is fixed.

nielsonm236, 2021-05-01:
I just verified with my development build and this anomaly does not appear there. I made some changes to my conversion algorithm so that likely provided the fix. The release should go out soon.

jmcvieira1, 2021-05-01:
Thank´s awesome job!
Now that you are testing i stumble on another strange behaver test  disable all i/o port´s leave mqtt and DS18B20 turned on save and reboot, and the modulle will not connect to
the mqtt server, i have to enable one port to the module connect to the server again.

nielsonm236, 2021-05-01:
I'm testing and developing all the time ... but I didn't think to try the above. I'll open a new issue.

nielsonm236, 2021-05-09:
Fixed with release 20210509 2031

closed 2021-05-09 by nielsonm236