nielsonm236/NetMod-ServerApp #67: Change port state using SAVE button not working on IO control page

flexiti, 2021-05-07:
Code Revision 20210221 1826 - is ok, next version not  (hangs if I press this button)

flexiti, 2021-05-07:
Ps . eg. http://192.168.0.17/03 and /04 method is working

nielsonm236, 2021-05-08:
I will roll back a test device and check this in case I missed something, but I know this is working OK in the current version (20210413 1254).

flexiti, 2021-05-08:
I just compiled 20210413 1254 -  not ok, and roll back to 20210221 1826 - ok
I wonder how it works for you? :)

nielsonm236, 2021-05-08:
Since you indicated your are compiling your own, have you tried using the pre-compiled .stp and .sx files? That might give a clue whether the problem is due to a difference in compiler environment.
If the pre-compiled files give the same problem, can you tell me what OS you are running and the Browser type and revision?
Thanks
Mike

flexiti, 2021-05-09:
yours .sx and .stp  also not work (windows 10 64bit, Edge 90.0.818.56 - latest and Chrome 90.0.4430.93 )

nielsonm236, 2021-05-09:
Well, this is really odd. No one else is reporting an issue like this. Are you using REST and GUI commands at the same time? A bug I just fixed (but haven't released yet) addresses an issue where using REST commands from the GUI while also trying to use GUI radio buttons can cause confusing results. But if you ONLY use the IOControl page ON/OFF radio buttons (no REST commands at the same time) it should work just fine. Can you tell me more about what steps you take? And maybe send screen shots of your Configuration and IOControl pages to my email address nielsonm.projects@gmail.com

flexiti, 2021-05-10:
[image]
[image]

flexiti, 2021-05-10:
[image]
[image]

nielsonm236, 2021-05-10:
The main anomaly I see is that the Gateway address should match your IP address. Since you are using 192.168.0.17 as your Network Module address the Gateway address should be 192.168.0.1. If you can give that a try please let me know the result.

nielsonm236, 2021-05-11:
I should add that in my understanding of routers it shouldn't matter that the Gateway address upper octets don't match ... as the Gateway (or default router) address shouldn't be used unless you are expecting routing to the internet (not just your intranet). BUT - there was one user where this made a difference. So, I suspect it may be router dependent.
If that doesn't fix it the only other thing I can think of is to be sure that you have no conflicts with MAC and IP addresses.

flexiti, 2021-05-11:
"Gateway address should match your IP address" - 17 is no gate to internet, and if I routing localy eg. form 192.168.0.60 to 192.168.0.17 this should be irrelevant if the gate is set to 192.168.0.1  eg. my computer have also this gate and there is no problem with connection to other local ip addresses.
MAC - I also tested other addresses.
I will do the test you ask for but in my opinion it is not an anomalous setting ;)
...Tested - still the same

ps. I have a wrong entry there, it should be 192.168.0.1 not 1.1 but in this case it does not matter
I will do more tests in the evening

flexiti, 2021-05-11:
Remember that in the older version it works with these settings :)

gruda74, 2021-05-11:
I am facing with exactly the same issue:
Tried out both NetworkModule.sx and NetworkModule-Browser.sx of the latest version and faced with the same freezing issue when clicked the Save button on the IOControl page.

Version 20210221 1826 is working fine with exactly the same deploy process and same settings = No changes in MAC or IP address. Only enabling some pins as input and output.

jmcvieira1, 2021-05-11:
Tested the latest firmware on Win 10 x64 with fresh instal off Edge Versão 90.0.818.56 (64 bits) no problem the port´s change fine the retain work´s, could´t it bee a cache issue?

nielsonm236, 2021-05-11:
Maybe ... but right now I don't think so. I just upgraded Chrome to the latest and now it is showing the issue also. So, it has something to do with the latest versions of Chromium (used in both Chrome and Edge). But having said that, I don't think the browser is at fault. I think the latest versions of these browsers have exposed some pointer issue in my POST processing routine. I now have a test routine that is repeatable, and I see that the EEPROM is being overwritten with an incoming string from these browsers. That should not be possible unless my code has a defect. So, I am pursuing that.

nielsonm236, 2021-05-11:
OK - found it and fixed it. The problem was provoked by a change in the pre-amble in the Chromium based browsers, but the defect was in my POST processing code. I wasn't properly handling the case where a packet ended during the first POST component, and that POST component was continued in the next packet. The browser pre-amble change shifted the POST data just enough to make this packet boundary crossing occur in just the right place. I will get a new build out within an hour.

nielsonm236, 2021-05-11:
Fixed in 20210511 1317

nielsonm236, 2021-05-11:
Just published Release 20210511 1317 with the fix.
THANK YOU for finding this and helping me with clues. It ended up being a one line fix but took some time to figure out what was going on.
I will leave this issue open until I hear that you agree it works.

flexiti, 2021-05-11:
Works fine. Thanks