nielsonm236/NetMod-ServerApp #68: Request for A/D Converter Inputs

nielsonm236, 2021-05-08:
I've had a couple of users ask if the IO pins can be used for analog input. Unfortunately the hardware designer did not bring any of the STM8 analog inputs out to IO pins ... the objective was just a relay controller at the time the hardware was designed. So, analog input is not possible with the hardware as-is.

I did some research and found that there is a 4 channel analog to digital converter that could be attached externally. It requires an I2C interface so would require 2 IO pins to operate it. I found one source on Amazon selling for about $3, but most sellers want $7 to $14. It is a potential solution if there is enough interest. I am not sure it could be included in parallel with the DS18B20 device, so this is just exploratory at this time.

[image]

https://www.amazon.com/gp/product/B07QJX4HNP?pf_rd_r=MK9NDZVXP5T5S4K12JJW&pf_rd_p=5ae2c7f8-e0c6-4f35-9071-dc3240e894a8&pd_rd_r=c22460af-fc02-4b59-ace1-f7ba7f3cedc2&pd_rd_w=0HdQd&pd_rd_wg=H3huC&ref_=pd_gw_unk

Pelicans-git, 2024-03-06:
This is a really cool idea. Did you implement this?

nielsonm236, 2024-03-06:
This was not implemented, but possibly could be. It is very likely that if a build were created that included this device it would have to be one that includes the I2C EEPROM (ie, a UPG build) and might require that some other feature(s) be turned off. The reason is that some of the builds already consume all of the Flash space.
Assuming the driver to operate the device was only a few hundred bytes, the biggest problems remaining are: 1) The GUI to configure the device, and 2) Defining the method to deliver the data (only via GUI? via MQTT?). I never started the development because of lack of definition and the high probability I couldn't get it to fit.