nielsonm236/NetMod-ServerApp #73: Device becomes unresponsive after several hours of operation

nielsonm236, 2021-05-14:
A user asked if I had seen the Network Module going unresponsive after a couple of days of operation, and at the time the question was asked I had not seen that. But now I think I've seen it twice on my test bench, so I will be trying to figure out how to track this down. Cycling power puts it back online.

If anyone else is seeing this please let me know. I don't know if it is firmware or if the hardware is having an issue. My test modules have been "abused" with lots of reprogramming cycles (hundreds, if not low thousands of cycles), so I don't know if I am seeing the effects of wearout.

nielsonm236, 2021-05-14:
Well this is probably a false alarm. I am using two power modules, one for relays and one for the Network Module. I measured the voltages and the power on the Network Module was 5.42v ... too high. I've swapped out the power supply and so far the module looks stable. I will update after a few days of operation if no further issues occur.

caladeira, 2021-05-14:
I've been running version "20210511 1317" in four modules for a few days
now with several Home Assistant automations without issues.

In the past I had a similar problem but it was related to a bad connection
in the power connector.
I re-soldered it and the problem went away!

nielsonm236, 2021-05-15:
I've determined that there is some hardware problem with my bench test module. Even after changing power supplies it still fails at a random amount of time. I suspect Flash wearout, so I will replace it today and restart testing. Keep in mind this module has been reprogrammed at least 1000 times, and has had a lot of other "abuse".

nielsonm236, 2021-05-16:
I have diagnosed this as: Dumb mistake on my part.
You would think 40+ years experience would have known better.
I had rearranged my test bench wiring and used a breadboard so that I could easily alter configurations for the recent YouTube video update ... and I used a cheap Dupont ribbon to connect signals and power to the Network Module via the breadboard. Well, it turns out the Dupont wiring was insufficient for the current required by the Network Module, and I was losing about 0.3V on each leg of the power path to the Network Module. Low voltage (around 4.46V) on the Network Module was causing the module to reset at random intervals. And this explains why the problems all started during the past week.
So ... basic re-learned is "always check your voltages at your Network Module and Relays".
I've put my original module back in service on my test bench and will close this issue.

closed 2021-05-16 by nielsonm236