nielsonm236/NetMod-ServerApp #75: publish_pinstate_all Port status information in binary form should rather be in text

flexiti, 2021-05-22:
The original  function returns data in binary form, rather it should be in text form. The return value was also strange or wrong.
Maybe this version will be better?

_//------------------------------------------------------------------------------------------
void publish_pinstate_all(void)
{
  uint16_t k;
  unsigned char app_message[5];       // Stores the application message (the
                                      // payload) that will be sent in an
				      // MQTT message.
				      
  static const unsigned char HexTab[] = "0123456789ABCDEF";
 
  k=ON_OFF_word ^ Invert_word;;

  app_message[0] = HexTab[((k >>12) & 0xf)];
  app_message[1] = HexTab[((k >> 8) & 0xf)];
  app_message[2] = HexTab[((k >> 4) & 0xf)];
  app_message[3] = HexTab[(k & 0xf)];
  app_message[4] = '\0';

  strcpy(topic_base, devicetype);
  strcat(topic_base, stored_devicename);
  strcat(topic_base, "/state");

  // Queue publish message
  mqtt_publish(&mqttclient,
               topic_base,
	       app_message,
	       4,
	       MQTT_PUBLISH_QOS_0 | MQTT_PUBLISH_RETAIN);
}_

nielsonm236, 2021-05-22:
Per the Manual returning the state as two bytes (16 bit binary) is per design. This was to minimize the size of this specific MQTT response.
Do you have a need for it to be in text form?

flexiti, 2021-05-22:
MQTT is text based , I think it should be plain text eg. "FFFF" or "1010101000000000"  or JSON data  eg. {"state":"FFFF"}  or {"state":"1010101000000000"}  .... but the decision is yours. I changed it. :)

flexiti, 2021-05-22:
PS. If there is still a binary version, check please, because it probably returns wrong data

nielsonm236, 2021-05-22:
MQTT is data-agnostic. You can put anything in the payload: test, images, etc. I can add an additional command to return the states in text form if really needed. Here's one reference with regard to data payload, but there are many:
https://www.hivemq.com/blog/mqtt-essentials-part-4-mqtt-publish-subscribe-unsubscribe/#:~:text=of%20the%20message.-,MQTT%20is%20data%2Dagnostic.,virtually%20every%20data%20in%20binary.
[image]

nielsonm236, 2021-05-22:
I will also add re-test of the existing function to my to-do list.