nielsonm236/NetMod-ServerApp #76: Allow changing name of relay states

AG7GN, 2021-07-28:
Thanks so much for this great firmware! 

In our environment, we wire devices we want to control to the normally closed (NC) contacts on the relays. So, when the relay is energized, it shuts off the power to the device wired to the relay. To avoid confusion for our users of the web interface, I was wondering if would be possible to make the relay state names editable via the web interface. So, if you are using the NC contacts, ON becomes OFF and OFF becomes ON.

nielsonm236, 2021-07-28:
Try the "Invert" checkbox with each pin. I think it does what you want ... assuming I understand.

AG7GN, 2021-07-28:
Thanks for the quick reply. I'm already using the Invert checkbox. I'll explain my situation in a bit more detail. The devices (radio and network equipment) I have wired to the relay board are powered on all the time. It's only when they are locked up and can't be controlled by any other means that I power cycle them using the relay board. I use the handy Timer feature in your firmware for that purpose, removing power for 30 seconds and then powering them back up. 

The relay board's and the HW-584's 5VDC power source has an inline fuse, as is our policy. Say that fuse blows. I lose power to the relay board and the HW-584 and anything wired to the relay board's *normally open* contacts are also going to lose power. For this reason, and because all of our equipment is normally powered on, we wire our equipment to the *normally closed* contacts of the relays so that even if the relay board has no power or there's some other problem with the HW-584, the equipment wired through those relays is still powered.

The web interface IO control states are ON and OFF. The ON and OFF describes the state of the relay "coil", not necessarily what's wired to the relay. That text works fine and is intuitive if your devices are wired to the normally open relay contacts, but not if they are wired to the normally closed contacts.  Hence, my desire to be able to change the relay state description (either a checkbox that inverts the ON/OFF state *text*, or select some other text. Most users are going to assume that ON/OFF is the state of the power for the device wired to the relay, not the state of the relay coil itself.

Thanks again for this great firmware.

nielsonm236, 2021-07-29:
OK - I'm in the middle of a different issue and it may be a couple of weeks before I get back to this. But I will definitely come back to it.
Mike

AG7GN, 2021-07-29:
No hurry, Mike. Thanks for considering it.

Steve

nielsonm236, 2021-08-16:
I'm trying to understand this issue and creating a drawing to explain (mostly to myself). Are your relays energized with a low logic signal, or with a high logic signal?

AG7GN, 2021-08-17:
Mike,

I believe the answer is energized with a high logic symbol. The bottom line is that when the relay is not energized, I'd like the label to reflect the actual state of the device wired to the relay's NC contacts rather than the state of the relay itself. In my case, the label should be "On" when the relay is not energized and "Off" when it is energized.

Steve

nielsonm236, 2021-08-17:
Take a look at the attached diagram. I think the current design does what you want. Let me know.
Mike
[image]

nielsonm236, 2021-08-17:
Let me know if you see a problem with my description or if you don't think it is working the way I described.

AG7GN, 2021-08-17:
Mike,

OK, I got it working the way I want it to, but not quite in the way you describe. This works for me:

Equipment that is normally powered on is wired to the NC contacts on the relay.

Invert: UNCHECKED
Boot state: ON

This does what I want: Even if the relay board is powered off, the devices wired through the relay's NC contacts are still powered. 

Thanks for working through this scenario with me.

-Steve

nielsonm236, 2021-08-17:
And it shows "ON" in the GUI when the equipment is powered ON?

AG7GN, 2021-08-17:
That's correct. Works exactly as I need it to.

nielsonm236, 2021-08-17:
Good. The implication is that the relay actually requires a logic low at the connector on the Network Module to activate the relay coil.

Given what you told me about your overall design and reliability concerns the one issue is the scenario where the Network Module loses power, but the relays still have power. In that scenario the Network Module will look like a logic low signal, causing the relays to activate and turn off your equipment. So, you'll need to assure that the Network Module and Relays share a power source (so they both lose power at the same time), or the Network Module has a very reliable power source so it can't power down when your relays are still powered.
Mike

AG7GN, 2021-08-17:
Mike,

We’ll be doing both: A reliable, battery-backed-up power source and the module and relays will share the same source.

Thanks again, 

Steve

nielsonm236, 2021-08-18:
Sounds like the ticket! Let me know if you have any issues. I will be doing another release shortly once I get documentation updated.
Mike

closed 2021-08-18 by nielsonm236