nielsonm236/NetMod-ServerApp #83: /98 /99 command does not return the correct size for the page in the HTML header

nielsonm236, 2021-08-16:
The title explains the issue. This is fixed and will be in the next release.

yozik04, 2021-08-17:
I am thinking if it would be easier to use chunked transfer encoding for html delivery See https://greenbytes.de/tech/webdav/rfc2616.html#transfer.codings

Jevgeni

nielsonm236, 2021-08-17:
Perhaps if I had started out that way :-)
For now I have fixed the rather simple math error for this specific case, and will add "chunked" to my think-about-it list. While pre-calculation of the total length of the transfer is a bit messy, once the code exists it is possibly even a bigger mess to back it out. There are days I feel like I'm working in a maze.

yozik04, 2021-08-17:
You do not need to precalculate anything. You will replace couple first bytes after your chunk is ready.

Jevgeni

nielsonm236, 2021-08-21:
Fixed in release 20210821 1541

closed 2021-08-21 by nielsonm236