nielsonm236/NetMod-ServerApp #89: Network Module project file

popk, 2021-11-11:
Hi nielsonm,

I want to try build your code. from your document say copy  NetworkModule.prjsm8  to my dir, I cannot find  this file.

can you provide this file ?

Thank you very much

pop k.

nielsonm236, 2021-11-11:
Hi - If you download the Source Code zip or tar file and expand it you will find the prjsm8 file included along with all the source code. I just checked and it shows up for me. Let me know if you still don't see it.
Mike

nielsonm236, 2021-11-11:
....

popk, 2021-11-12:
hi Mike,
sorry , .prjsm8 on my computer was associated with microsoft outlook , so I cannot see it.
anyway what program should it run to this file?   stvd  , stvdebug.  I did not see any program use it.
pop k.

nielsonm236, 2021-11-12:
While STVD can be used (I could not get it to work right), I've used IdeaSTM8 in all my development. If you followed the instructions in the manual in the section titled "Developers: Setting Up a Development Environment" the IdeaSTM8 program will have been installed. On my system this automatically associated .prjsm8 files with the IdeaSTM8.exe program.

The instructions in the manual should have created a directory named "C:/Program Files (x86)/COSMIC/FSE_Compilers/CXSTM8" and the IdeaSTM8.exe file will me located in that directory.

If for some reason the file association didn't work you can associate the NetworkModule.prjsm8 file directly with IdeaSTM8.exe.

Mike

popk, 2021-11-12:
Thanks for you kind.
 Now I can do it 
thanks you very much

pop k

closed 2021-11-12 by popk