nielsonm236/NetMod-ServerApp #90: Always ON 16 channel relay modules

michalwoz, 2021-11-23:
I have problem with 16 relay module which is precisely described in documentation. Opto-isolator remains activate regardless of the state of the output pin on the Network Module. Changing pin mode from output to input dis-activates opto-isolator. In AVR uC it was possible to change OUTPUT state by changing pin direction. Is it possible to implement software feature to configure:
"ON" as: pin is OUTPUT state is LOW
"OFF" as: pin is input
?

nielsonm236, 2021-11-23:
I'm checking some previous notes when I helped another user fix this issue. I will get back to you shortly.

michalwoz, 2021-11-24:
@nielsonm236 From hardware perspective i think that it would be easier for me to replace LM2596 from 5V version to 3,3V on 12V relay board and short circuit AMS 1117 on Network module than adding extra diodes on path from network module to relay module.

Software solution would be perfect.

On AVR we have:
[image]

I've used row 4 to get LOW state and relay ON.
and row one or three to get relay OFF.

Do you think that something like that can be implemented?

nielsonm236, 2021-11-24:
There is a simpler fix with a single diode fix on the Relay board. I will get the information to you shortly. The problem is that another some additional voltage drop is needed to make the optos turn off, and that can be done with one diode at the output of the 5V regulator. I have a writeup but need to attend to a family issue first. So it will be another couple hours before I can get that to you.

michalwoz, 2021-11-24:
Sorry, it wasn't my intention to hurry you up. Take your time.

In the meantime I've looked into STM GPIO. Isn't it my problem that P-BUFFER is enabled when my relay should be OFF...?
Maybe we don't have to switch to input mode to switch off the relay, maybe we can just set CR1 to 0?
https://github.com/nielsonm236/NetMod-ServerApp/blob/8b76cbf555f8da93b1ac341af2f86b936f792046/NetworkModule/Gpio.c#L205

If you think that it may work it would love to test it.

[image]
[image]

nielsonm236, 2021-11-24:
Re: "Isn't it my problem that P-BUFFER is enabled when my relay should be OFF...?"  No, the problem is a little more subtle than that. The problem is the current flow through the protection diode connected to Vdd (3V) and, in parallel, the small amount of current flow through the pullup resistor when the output pin has a voltage on it that is greater than 3.3V.

When you change the pin to "Input" the pull up resistor is eliminated, but the protection diode is still there. This small amount of current change is leading you to believe you fixed it ... but very small variations in voltage and load will cause some of the relays to operate intermittently. So, the software approach will still lead to some disappointment.

In my manual I suggested a diode in series with each pin to drop additional voltage, but I agree that is a very messy solution. Working with another user we came up with a modification for his 16 relay board that involved cutting one trace and inserting a diode in series with the 5V power that supplies all the Opto isolators. It worked great for him. But, in that investigation we found that there are 4 or 5 different board layouts, so it takes some careful observation to cut the right trace. Not difficult, but it does take a few minutes to figure it out on your specific relay board. I've attached the write up explaining how to do this. The attachment is PDF ... let me know if you prefer another format.
[16 Channel relay board modifications.pdf](https://github.com/nielsonm236/NetMod-ServerApp/files/7592105/16.Channel.relay.board.modifications.pdf)

nielsonm236, 2021-11-24:
I should also add that your suggestion "... replace LM2596 from 5V version to 3,3V on 12V relay board and short circuit AMS 1117 on Network module ..." would also work. But one caution: You may not want to use a single power source. Several people have had problems with voltage glitches on the power supply for the relays, and when they used that same power to provide power to the Network Module they found it would reset intermittently. So it is very important to keep the Network Module power as clean and stable as possible. In all my implementations I've gone so far as to use separate power supplies for the Network Module and the relays. But some have reported success using a single supply. I think the hardware reset circuit within the STM8 is a bit too sensitive.

nielsonm236, 2021-11-24:
And another afterthought: If you use my "cut a trace and add a diode" approach you may have to do a little more to make sure you still have 5V on the connector going to the Network Module ... IF you are going to try to run the Network Module from the Relay module 5V.

michalwoz, 2022-01-23:
Today I've replaced LM2596 to 3.3V version. Everything works.

nielsonm236, 2022-01-23:
A quick additional question before we fully close. By replacing the LM2596 the 5V pin on the header of the relay module will now have 3.3V. This will conflict with the 5V on the Network Module pins. Did you do something additional to eliminate that connection?

michalwoz, 2022-01-23:
> You may not want to use a single power source

I've followed your suggestion and will use different power sources for relay module (12V) and different for LAN module (5V)
I've just put red label with "3.3V" on relay module and defaced "+5V" on connector description. I will not use VCC connection between modules. You may consider adding this solution do fixes for 16 relay module and close this issue. Thank you for your help.

nielsonm236, 2022-01-23:
OK - that makes sense. Yes, I will add this solution to the manual for others to reference. Your solution is a great idea as long as the user makes sure they don't end up connecting the 3.3V on the relay module to the 5V on the Network Module. This is actually the best solution I've seen ... but of course requires some surface mount soldering ... not super fine so very doable.

closed 2022-01-23 by nielsonm236