nielsonm236/NetMod-ServerApp #91: Major problem: Latest Browser releases not working with the 20210821 release

nielsonm236, 2022-01-14:
A user has reported that the 20210821 release causes pin definitions to reset to "disabled" when pin state changes are made in the IOControl screen. The user reports that the 20210511 release works OK.

With some testing at my location I've confirmed this symptom in Chrome and Edge, both of which are based on the Chromium engine. This specific symptom does not appear in Firefox, HOWEVER, I see that the "IOControl" button has disappeared on the Configuration page when using Firefox.

As an FYI, I dealt with a similar bug the last time Chromium was updated. Based on that experience this IS NOT a bug in Chromium, It is likely a bug in my code where I parse incoming browser packets. The last time the problem appeared I found that I was not handling all browser packet boundaries properly, and my guess is that I still have a problem there.  Because of limited memory packet boundary processing is a bit convoluted. So, it is a place to start.

My recommendation while I work on this:
a) This probably does not affect MQTT users (until you try to use the Browser for access).
b) Use the 20210511 release if you are having Browser problems. Unfortunately going back to that code takes you back to code that does not support online upgrade. So, I must fix this quickly. See the next recommendation.
c) Use Firefox as your Browser. This is only a partial fix as the Configuration page bug is not showing the IOControl button and you will have to use the IPAddress/60 URL to get to your IOControl page. But at least you can continue using the 20210821 code.

I'M WORKING ON IT!   
Mike

gabortamasko, 2022-01-14:
Thanks! Keep us posted!

nielsonm236, 2022-01-17:
I have this resolved. I no longer believe it is related to Browser updates despite it having symptoms that looked like a previous "Browser provoked" bug. The problem is a very basic bug in my code that handles POST data (the data sent from the Browser when you click Save). This bug has been there for a very long time, so I think it may affect several releases even if you haven't noticed it in prior releases. Essentially what happens is that a Save on a IOControl page is interpreted as coming from the Configuration page, and since the IOControl page does not have any of the Configuration data all the Configuration data is returned to the default state. It is the result of my misunderstanding some of the obfuscations in the melding of source code from multiple sources.

The fix is fairly simple: Once I understood what was happening I added code to verify the POST source (IOControl or Configuration) before making any updates to the parameters.

I need to do some more testing, then write up the documentation, then I will release. If there are any volunteers that are willing to give the new code a try before I release it please let me know. I'll need to know what variant of the code you want to try (Browser / MQTT / Online Upgradeable). My initial testing has been on the Online Upgradeable Browser code.

Mike

gabortamasko, 2022-01-17:
I will test it this week or so. Thanks

nielsonm236, 2022-01-17:
Let me put a "Beta" code set up on GitHub. I haven't done that before so it will take me a bit to figure out how that is done.

nielsonm236, 2022-01-18:
"gabortamasko" and Jose - I know this isn't the "right" way to do this, but my attempts to understand how to make a Beta available on GitHub have left me concerned I might inadvertently mess up my Master repository. So, I've built the needed files for test and will attach them here. As FYI, my branch with the fixes is the "code-fixes" branch. But you shouldn't need to access the branch, just the files attached here
[NetworkModule.zip](https://github.com/nielsonm236/NetMod-ServerApp/files/7885665/NetworkModule.zip)
.

nielsonm236, 2022-01-18:
I've been running tests on all four variations of the firmware and so far it looks good to me. Let me know if you see anything.

jmcvieira1, 2022-01-19:
Hello, I updated two modules with the MQTT/UPG version and so far they are working correctly, if there is any problem I will let you know.
Thank´s

Musicman50, 2022-01-22:
Hi Michael. Thanks for this update. I hope to be trying the new beta code this week sometime.

nielsonm236, 2022-01-22:
Thank you. I'm anxious to get feedback as this particular bug shook my confidence in my own testing. And, of course, I'd like to get a fix out to the larger group of users. It looks very stable to me running on 10 units with a mix of Browser Only, MQTT, with and without online upgrade capability. But, like I said, this was a pretty significant bug and I missed it when I tested that last official release.

nielsonm236, 2022-02-03:
I'm not seeing any negative feedback and my testing is all positive, so I plan to make a general release this weekend. After that I will go on to implementing the BME280.

nielsonm236, 2022-02-05:
Addressed in Code Revision 20220205 1645

closed 2022-02-05 by nielsonm236