nielsonm236/NetMod-ServerApp #93: Add BME280 Temperature Humidity Pressure sensor

nielsonm236, 2022-02-05:
Issue #61 started a discussion on adding multi-function environment sensors. I've settled on the BME280 as the right one to add. This may also allow support of the BMP280 sensor (temperature and air pressure only). If you are trying to buy the BME280 you need to be very careful about your vendor. Many show (and describe and charge for) the BME280, but ship the cheaper BMP280 to you. You can tell right away: the BME280 "can" is square in shape. The BMP is rectangular.

I will start work on this code right away and will post progress. I suspect it is only going to fit in code releases that take advantage of the externally added I2C 24xx1025 EEPROM.

nielsonm236, 2022-06-20:
Well, I worked on this code for awhile and thoroughly messed it up. I may have to dispose of the branch I was working on and start over to figure out what I did wrong.
I'm temporarily indisposed due to family issues. Not sure when I will get back to this as this BME280 code is much more complex than I expected.

sjorsBe, 2022-10-10:
Hi Michael, 
sorry to read about your family issues!
Whish you all the best 

a small question though,  whats the speed of the i2c bus thats being used with the netmodule?

kr 

Sjors

nielsonm236, 2022-10-10:
The I2C bus is a "bit bang" implementation on two of the IO pins. I haven't actually measured the speed. I will set up my scope again some time and get an estimate. It doesn't vary much as any time the bus is operated it is in uninterupted straight line code.

nielsonm236, 2022-12-06:
It's been over 9 months but I'm going to take another run at this. The example firmware is very complex to translate into a bare metal platform. I will start over on it perhaps implementing just one function at a time.

nielsonm236, 2022-12-18:
I'm working on this now. I will scrap my previous effort as it seems to be seriously broken and more difficult to fix than to start over. Biggest problem with this addition is the RAM needed. Looks like at least 100 bytes of RAM. Since it will be targeted only at Upgradeable firmware implementations the Flash needed is not a problem even though it is probably up to 3000 bytes.
To get RAM I will probably have to create a new build type that removes the DS18B20 code and utilizes the BME280 instead. So, you'll have to select a DS18B20 build OR select a BME280 build. Or maybe I can have a runtime memory allocation that creates a structure for DS18B20 (about 60 bytes) or creates a structure for BME280 (about 100 bytes) based on feature bits. Either approach is very code intrusive so it will take awhile to implement. I think @jmcvieira1 is the most interested in this addition, but I can also utilize the functionality. Comments welcome.

jmcvieira1, 2022-12-18:
hello the BME280 already have a temperature sensor i´m ok on leaving the DS18B20 out of the build if there is no space or is more difficult to implement the code for it.
Thank´s

nielsonm236, 2022-12-18:
This will be a RAM issue, so I was looking for a way to free up RAM. Removing the DS18B20 functionality when the BME280 is present made the most sense for exactly the reason you mention.

nielsonm236, 2022-12-21:
Two quick updates:
1) Answering @sjorsBe question: The I2C bus operates at 100KHz. Not fast, but not bad for a bit-bang approach on this processor.
2) I built code for the BME280. It turned out to be about 4800 bytes. I can optimize once I get all functions to work, but I will be surprised if I can get more than about 800 bytes out of it. So, this suggests there will be a separate release for the BME280, and future updates to that release will probably only include bug fixes (not added features). Too early for me to tell for sure. PROBLEM: my work has stopped as I've hit some kind of bug in the IdeaSTM8 tool set. I have a question in to the Community to figure out how to get it running again.

nielsonm236, 2023-01-14:
Update: I've been working on this for about 7 weeks now. Sounds like a long time but even in retirement it is amazing how little time there is for the fun stuff.
Good stuff:
- The driver works, and I can collect Temperature, Pressure, and Humidity data. Right now display is only via the Browser.
- I've implemented this with define statements so it can be included/excluded.
Bad stuff:
- The driver takes a large amount of Flash, even after significant "code compression". The BME280 was definitely not designed for tiny environments.
- I've determined that BME280 support will require its own build as Linked pin support and DS18B20 support must be disabled to allow room for the BME280 driver. There will still be room for future bug fixes (I hope) but BME280 support will be a bit of a stand-alone release.
- Using the UART interface for development debug is very limited due to Flash space constraints.
- The code is getting very cluttered with defines for including/excluding the various features. Makes things much more prone to development errors.
Remaining development:
- MQTT and Home Assistant interface for the BME280. Hope it fits.
- GUI Javascript for "BME280" checkbox.

nielsonm236, 2023-01-22:
This is working now, and I have updated the Manual to describe how to use it. The BME280 driver is very large compared to the modest Flash space, so it will be a special build with each release. That special build will have some limitation, for instance no Linked pins if you want to use the BME280.

I have two BME280 devices and connected each to a different Network Module to perform comparisons of Temperature, Pressure, and Humidity. I found that Pressure is within 1 hPa between the modules, however, Temperature is 1 to 2 degrees C different between modules, and Humidity is about 5% points different between modules (for instance 21% on one module, and 26% on the other).

I some online research with regard to temperature and humidity accuracy for the BME280. That did not give me much confidence that I should expect temperature or humidity accuracy, and in fact just using my two sensors if the "real temperature and humidity" is half way between what I see on the two sensors I have, then both sensors are within the expected tolerance of the device. I find it unlikely that I managed to buy two sensors that are both at the outer edge of the tolerance (one high and one low), but without a lab standard to compare them with I don't have much to argue with.

As FYI, of my two sensors the one that is measuring "high" on temperature compares very closely with the temperatures measured by DS18B20 sensors on some other modules. But that would imply that the BME that is "low" on temperature is about 2X out of tolerance. Still not much to argue with, but interesting. And of course I bought at the cheapest price I could find, so maybe the manufacturer was sloppy about post-assembly baking, thus causing tolerance issues.

For now this will have to do, as it is the same code running both sensors, leaving me with the conclusion that the measurement differences are due to the sensor itself. With some field exposure maybe there will be further findings and education that will help.

At present the BME build only has about 350 bytes to play with if code adjustments are needed.

The next release (pending) will include this code. You won't see any differences unless you install the BME280 build.

nielsonm236, 2023-01-27:
Addressed in release 20230125 2003.

closed 2023-01-27 by nielsonm236