nielsonm236/NetMod-ServerApp #95: MQTT state-req returning all zeroes

Harry-Burnham, 2022-02-16:
Hi, newbie here, so I may be missing something obvious, but when I issue the state-req command the return message payload appears to be zeroes. I use Mosquitto pub on a PC to issue the commands and mosquitto sub in another command window to view the return, If I issue a state-req command I simply get **NetworkModule/HeatControl/state** shown with no payload following. If I use the Format option with %X to show the payload in Hex, I get 0000.  It shows the correct hex values for ON and OFF with set commands.

In order to establish whether there was a problem in the mosquitto sub output I sniffed the network packets and the bytes where I expected to see the payload were definitely both zero.

I have installed the new version of the firmware which shows as "code revision 20220205 1645" on the browser configuration page. I get the same results with this. It doesn't seem to matter what relays I set on, whether they are set on from the browser page or MQTT command, the return is still zeroes. The correct output always shows on the /99 browser page, though.

I have tried searching the issues already reported but I can not find any mention of this problem. Any help will be greatly appreciated. Thanks, Harry

nielsonm236, 2022-02-16:
I just tried with MQTTLens and I see the same thing ... bytes 0 and 0 when I should see something else. I will look into this.

nielsonm236, 2022-02-17:
WELL ... that's embarrassing. A very first grade error in a  for() loop. This was tested early on and working, but I suspect I broke it in subsequent code reduction work and failed to test it. In any event, it is now found, fixed, and tested. This specific feature must not be used much as I haven't heard from anyone else.

Is this critical to your application? I'm working on another code addition and would like to wait on a release if possible. But if you need it right away I can send you a special build.

I greatly appreciate your bug report.

Mike

Harry-Burnham, 2022-02-17:
well, I'm just glad that it wasn't me doing something silly. I am gradually getting to grips with the ways of using MQTT amidst all the other stuff that has to be set up for a smart home, and finding the odd thing that I have misunderstood.
It's not critical in that I can obtain the same information via the /99 URL, so I can use that until your new release - it is only a matter of my own sense of economy of code to use MQTT for everything rather than MQTT and the REST commands in the same application, so I'm quite happy to wait until the new release comes out. 
Thanks for a great piece of software, it's really making my project a lot easier to get off the ground.
Harry

nielsonm236, 2022-02-17:
OK - Thanks. I will leave the Issue report active until the fix is released. Feel free to report anything you see that is amiss, or even if it just needs a better explanation in the manual.

nielsonm236, 2022-08-17:
Just had another request for this. I'm going to get a build out with this fix.

Harry-Burnham, 2022-08-17:
OK, that's great, thanks.

nielsonm236, 2022-08-17:
Arrrgh! My compiler license expired. Will take a couple of days to get it updated.

nielsonm236, 2022-08-20:
It's been awhile since I've done a release so hopefully I didn't screw this up. I only fixed issue #95 for this release (see the 20220817 1508 General Release).
Please check it out and let me know if you see anything of concern.
Thanks
Mike

Harry-Burnham, 2022-08-21:
Hi, thanks for that. I will download this release but it may be a while before I can test it as that particular project - central heating controller - is on the back burner during the fine weather as I need to get the garden landscaped. Thanks, Harry

nielsonm236, 2022-08-24:
By email I received a reply that this fix appears to be working properly. I'll close this one and move on to a couple of others.

closed 2022-08-24 by nielsonm236