nielsonm236/NetMod-ServerApp #96: Drilling 3rd hole in HW-584, how many layers PCB is?

michalwoz, 2022-03-31:
@nielsonm236 what do you think about drilling 3rd hole in HW-584?
I can see only GND on both sides of PCB (screen). How many layers PCB is? I think that only 2.
I need one more point to fit PCB. Fitting it by only 2 screws is not stable enough.
[image]

nielsonm236, 2022-03-31:
I didn't design the hardware so this is at your own risk. But, in all the "reverse engineering" that I did all signals were on the top and bottom layer. So I believe it is only two layers. I am not sure that top and bottom are both ground, but you may be able to use a meter to verify. If the point you drill does not have the same power plane on top and bottom you'll have to be careful that your hardware doesn't short the planes. Maybe use a nylon screw and nut if you aren't sure.
Hope this helps.  Mike

michalwoz, 2022-03-31:
I hope that few dozens od little holes in PCB are "vias" which connects GND layer on both sides of PCB. That needs to be verified.

nielsonm236, 2022-03-31:
Yes - check that carefully. Most of the vias are signal traces. If you don't see a trace connected to the via on either side then I think you are correct. You might try scraping away the silkscreen on both side where you plan to drill and check with a meter.

nielsonm236, 2022-04-27:
I'm closing this one.
michalwoz - open a new one if more assistance is needed.

closed 2022-04-27 by nielsonm236