nielsonm236/NetMod-ServerApp #97: getting a bunch of quetionmarks

TurboTronix, 2022-04-21:
After flashing the board and going to http://192.168.1.4:8080/60
I get a huge line of �������������

TurboTronix, 2022-04-21:
I think this means the board is dead...
I tried another board and all is good.

nielsonm236, 2022-04-21:
You might try re-flashing the suspect board. I've found these are really hard to kill, with one exception: Reverse power will almost instantly destroy the on-board voltage regulator, and power greater than 5.5v will slowly kill the regulator. But if you have some soldering skills it is not too hard to replace.
However, if you are seeing the question marks as you show here it suggests the board is trying to respond. I have seen this before, and I think (recollection is vague) that it was a browser problem. I can't recall if it was a down level browser or if it was IE (which is not supported).
So before you ditch the suspect board: 1) Check 5V power input, 2) Check 3.3V on the output of the regulator, 3) Try reflashing, 4) Try an up to date Chrome or Firefox.
Let me know how it goes ... your reply might provoke another suggestion.
Mike

TurboTronix, 2022-04-21:
I do have soldering skills but lack the knowledge of identifying which chip is which.

nielsonm236, 2022-04-21:
I attached some illustrations for the regulator. Don't replace it until you measure the output voltage. If you see 3.3V the regulator is still good. They are cheap - shipping will probably cost more than the part. I've read some people have trouble with Chinese knock-offs ... I don't know if that is true.
[image]
[image]

nielsonm236, 2022-04-21:
Are you in the US? If yes and you need to replace the regulator I might be able to save you some money. Send an email to my project address nielsonm.projects@gmail.com.

jmcvieira1, 2022-04-21:
Hello I already had these question marks in the browser when I was reprogramming a board recently and I solved it by uploading the firmware again with the usb stick.

TurboTronix, 2022-04-21:
I do get 3.3V so I guess that is not the issue. I reprogrammed several times already but always get the exclamation string.

jmcvieira1, 2022-04-21:
Now I remember what happened, the board had the firmware NOT upgradable and I used the usb stick to upgrade to the upgradable firmware, but did not follow the instructions and flashed the board with the NetworkModule-MQTT-UPG.sx file, and after putting the board up and running on the network I was greeted with question marks.

TurboTronix, 2022-04-21:
Ok so I was uploading the MQTT-UPG and now that I've uploaded the MQTT (not UPG) it seems to work.

jmcvieira1, 2022-04-21:
These boards are very tough as far as firmware is concerned as Mike mentioned earlier, I still haven't been able to damage any.

nielsonm236, 2022-04-21:
Jose has it right. The UPG firmware requires the added external EEPROM to work. If that is not installed then the firmware can't find the webpage images. The non-UPG firmware has all the webpages integrated on the processor flash.

nielsonm236, 2022-04-27:
I'm calling this one resolved since it boiled down to using upgradeable code without the external EEPROM (a requirement). Good discussion though.

closed 2022-04-27 by nielsonm236