nielsonm236/NetMod-ServerApp #98: Reset button sets default web port to 8080

yozik04, 2022-05-12:
I have flashed the latest MQTT version (without upgrade functionality)

Just a place for discussion. Why port 8080 but not 80, which is default http port?

nielsonm236, 2022-05-13:
The reset button resets a LOT of things to defaults, including the IP Address. Now ... why in the world did I make 8080 the default port. I know I had a reason at the time but can't remember what it was (perhaps not a good reason). 8080 is also a common html port number.

I suppose future releases could use a default of 80. I would just need to update the docs.
Mike

caladeira, 2022-05-13:
I always assumed that port 8080 was for historical reasons.

As far as I can remember, the original firmware that came with these boards
used port 8080.

In my modules, I've changed them to port 80.

nielsonm236, 2022-05-13:
I'm pretty sure the originals used port 80, because you didn't have to indicate a port number in the URL (other than the fake "port" they describe). I'm going to dig through my backup modules to see if any are still original. I don't think they are but maybe I'll get lucky.

I'm starting to wonder if I made the default port 8080 to make sure people understood / used actual port numbers. If so that was a pretty dumb reason to do it. I'll look back through some ancient notes to see if I can find my reasoning. I know that on lots of other equipment where I'm doing port forwarding through my router I use alternate html port numbers ... maybe that's what I had in mind. But I could easily make future releases use 80 as the default. Certainly less typing and forgetfulness when bringing up a new module.

nielsonm236, 2022-05-13:
As soon as I hit "enter" I had a vague recollection. When starting with a raw module from the factory it uses 192.168.1.4 port 80. When you program my code I wanted to make sure that your first contact with the module was assured to be the new firmware, so I made the default 192.168.1.4 port 8080. This way you knew for sure that the new firmware was there as it no longer responded to port 80, but instead to port 8080. Now, that's not a really good reason for doing it, but when my firmware was much simpler (and I was still learning how to make it work at all) this was a good first test. I think I might has also been encountering problems with browser caching, and changing the port number helped with that.

nielsonm236, 2022-09-22:
Based on the previous comment and the fact that I have it documented in the Manual I'm going to close this one. Re-open if there are any really big concerns.
Mike

closed 2022-09-22 by nielsonm236