nielsonm236/NetMod-ServerApp #99: Data Memory corruption when you enable DS18B20 and configure outputs at the same time

yozik04, 2022-05-12:
Latest MQTT fw.

I have managed to repro this at least 3 times in the row.

When you configure device first time:
Try to fill all fields at once and put DS18B20 flag as well. You will get memory corrupt.

Configuration I attempted:
[image]

Save responds with zero byte length and no redirect is triggered. Device stops responding after that.

I forgot to make a screenshot from STVP's Data Memory tab. But it gets corrupt starting from the first byte.

nielsonm236, 2022-05-13:
Just to give me a few more clues: Does the corruption happen if DS18B20 is NOT enabled for this Save?
I'm very surprised we would get a total EEPROM corruption under any condition ... but I've been surprised before.
Mike

yozik04, 2022-05-13:
Hmm. I am unable to reproduce that with my other module. Even by sending the same payload (I have saved that previously).

I'll close this for now. Maybe it is because I have a lot of stuff attached to the first unit already.

closed 2022-05-13 by yozik04