nielsonm236/NetMod-ServerApp #100: Remove Configuration Button

Force-GT, 2022-06-19:
Can you please make a webpage that doesn't have the "configuration" button? There is always some user that likes to mess with the settings. 
Thank you

nielsonm236, 2022-06-20:
I'm guessing here you are an admin in some multi-user environment. This really wasn't meant for multi-user, but I think it would be easy enough to produce a version that didn't have a Configuration button. There is no password capability, but access to the Configuration page would require knowledge of the URL page code to open the Configuration page. That's fairly easy to hack, but assuming a fairly benevolent environment it might work. Does this sound adequate?

nielsonm236, 2022-06-20:
I should add that by "produce a version" I meant a special code load that provides this capability. There isn't Flash space in the versions without the added external EEPROM to add another webpage. So, I would create a version that you could load that provides this ability.
Now, if you happen to have added the external EEPROM there is quite a bit of space for added web pages.

nielsonm236, 2022-06-20:
Let's define a little:
Would the new release work as it does now? With the Configuration button visible and working?
If yes, then when someone sets up a module they would click a "disable Configuration button" checkbox (presumably on the Configuration page). From that point forward the Configuration button would no longer work on the IO Control page. BUT, the admin knows they can still access the Configuration page with http://IP:Port/61. This only makes it slightly less accessible to a "non admin" user.
Feedback?

jmcvieira1, 2022-06-20:
Exactly, that's right, the average user won't accidentally get to the configuration menu, which has already happened here at my home.

Force-GT, 2022-06-21:
Yes Michael, that would be perfect!

nielsonm236, 2022-09-02:
I've started looking at this one. The idea is not complicated, but it will require some javascript and use of configuration data in the IOControl webpage that is not currently used there. So "bytes consumed" will be the primary issue. I'll try some experiments over the next few days.

nielsonm236, 2022-09-11:
I'm testing this now. So far it looks good. Will put out a Pre-Release in a day or two.

nielsonm236, 2022-09-11:
I published a pre-release to address this issue. Let me know if you see any problems.

closed 2022-09-22 by nielsonm236