nielsonm236/NetMod-ServerApp #102: Help Requested setting up Homebridge / MQTT-Thing on Raspberry Pi

nielsonm236, 2022-07-30:
Two users (H.K. de Koning and Sjors Beens) asked almost the identical question within minutes of each other today. Unfortunately I don't have the experience to answer their call. The basic question: Can you help me figure out how to set up MQTT-Thing to interface with the Network Module?

My comments to H.K. and Sjors: I'm using Mosquitto running under Windows 10 as my broker. And I use Home Assistant to manage devices. I personally am not a MQTT expert when it come to setting up the broker management - I too needed help from others to get that to work. Now, of course, you are trying to use different tools on Raspberry Pi, which I think is great because of the low power nature of that device.

A question for you: Have you been able to interface any other device to your Homebridge / MQTT-Thing configuration? Or is the problem even getting those two software components working on Raspberry Pi? Just trying to figure out if you can't talk to the Network Module, or if you haven't been able to talk to ANY device so far.

Does anyone else our there have any experience with this that might help out these new users?

sjorsBe, 2022-08-03:
Hi all, 
Since the above question is mine, i would like to let you guys know what the progress is on the topic. 

**The problem:** 
I would like to control the pins on a netmodule through homekit. 
in order to have a good interface to homekit I installed Homebridge on a Raspberry Pi.
Quite a lot of home automation goes through the Homebridge. 

In order to use the netmodule i reflashed the firmware with the MQTT version. 
But to be honest: MQTT was quite new to me. 

So I ended up with 5 issues: 
1. How to install a MQTT broker on a raspberry Pi
2. How to connect it with the Homebridge solution 
3.. How to configure the Netmodule so it can communicate with the broker 
4. How to configure the broker to understand its clients 
5. How do I understand the strange MQTT principles and language.....

The installation of a MQTT broker on a Raspberry PI and to connect it with the Homebridge was quite easy:
Homebridge has the possibility to install plugins, One of its plugins was according to all the documentation on the net, the best and most versatile  MQTT broker available: it is called MQTTThing foe Homebridge. 
That took care of issue 1 and 2.
But it took me a heck of a lot of studying to sort out how the MQTT protocol was put together and what dialect was used for MQTTThing.......
Hence my question above: How should I configure the Netmodule and the broker to talk to eachother? It would have helped with issue 5.....

The Netmodule manual helped a bit. With a lot of study the remarks made more and more sense. (i dont grasp a numer of them yet but time will tell :) )

After a lot of trail and error  it still didnt work. it kept telling me that I have a ECONNerror wich nobody could explain. Also the Netmodule only showed max 3 green dots on the configuration screen. Not good!

In the end I gave up. In a last effort i walked through all the available plugins for Homebridge and foud a tool called Aedes. It was described as a basic MQTT broker. So I just installed it. 
Suddenly everything worked fine! 5 green dots on the netmodule, the tracer tools did function well etc etc. 
It turned out that the bloody MQTTThing was not a broker at all but just a translater to a broker. 

So now i have it all working 
Frustrated but happy. 
In a next episode of this issue i will  inform you guys how to set it up properly. 
So you dont make the stupid mistake i made :)

Have fun!

Sjors

yozik04, 2022-08-03:
@sjorsBe You need to create a new user in MQTT Broker. Google will help how to do it. Then you will enter credentials of that new user in NetModule. Download MQTT Explorer to monitor what is being sent (you can reuse same credentials). When you will see NetModule topics in MQTT Explorer proceed to configuring MQTTThing.

sjorsBe, 2022-08-03:
mqtt broker is cloud based isnt it. 
i dont want to depend on the cloud for switching on my lights :)

yozik04, 2022-08-03:
Install mosquitto on RPI

nielsonm236, 2022-08-03:
Interesting. When I read up on MQTT-Thing it sure sounded like it was a broker ... not just a translator. But it appears you found out differently!
If you installed Aedes on your Raspberry Pi and now it works you don't need Mosquitto. But I know that Mosquitto is commonly used as a broker on Raspberry Pi and I am using it on Windows 10. I know others are using it in Linux environments. I know nothing of Aedes, but if you've got it working you should be OK.
MQTT brokers CAN be cloud based, but as you found you can have your own personal broker on your internal network. That's what I do.
By the way, it seems lately the term "MQTT Server" is the more accepted term for the broker ... but we all seem to keep saying "broker" anyway. :-)
You really shouldn't need to know much about how MQTT actually works. You only need to know how to set up a broker. Then if the devices are told what the broker IP address is things like the Network Module will just connect for you. Almost magic.

sjorsBe, 2022-08-03:
Eades comes as a direct plugin in homebridge. 
Makes life quite easy :)

nielsonm236, 2022-08-03:
FYI - I was pleased to see yozik04 join this conversation. He's very busy, but last year he helped me a great deal when developing the MQTT details on this device, particularly with regard to the Home Assistant MQTT interface. He can't always respond but he has very good input when he does.

yozik04, 2022-08-04:
Aedes is plugin for Homebridge allows you to host an embedded MQTT broker (server) within Homebridge. So you should be all set. It is not in the cloud.

yozik04, 2022-08-04:
> FYI - I was pleased to see yozik04 join this conversation. He's very busy, but last year he helped me a great deal when developing the MQTT details on this device, particularly with regard to the Home Assistant MQTT interface. He can't always respond but he has very good input when he does.

Thank you Michael. I am still subscribed to your great project. Feel free to ask if you need any help with some topics.
Thanks to your great work I am now able to control all my floor heating valves using these awesome modules.

nielsonm236, 2022-08-24:
I think we answered this one. I'll close it. Feel free to open a new issue if there are more questions.
Mike

closed 2022-08-24 by nielsonm236