nielsonm236/NetMod-ServerApp #104: Max current through an input or an output pin

sjorsBe, 2022-08-03:
Does anyone know? 
What is the max input pin current or output pin current an Netmodule can handle?

and how affects having all the 16 ports active a long time?

Thanks in advance 

Sjors

nielsonm236, 2022-08-03:
It's in the device spec, Section 9. I've attached here for reference. Read carefully, as there is not only a per pin spec but a "maximum for all pins" spec. You can't just multiply the per pin spec by the number of pins.
[STM8S005 Spec Sheet - DM00037613.pdf](https://github.com/nielsonm236/NetMod-ServerApp/files/9254343/STM8S005.Spec.Sheet.-.DM00037613.pdf)

nielsonm236, 2022-08-03:
Additional comment: One of the most common interfacing problems people have is connecting the Network Module to 5V logic. The processor operates at 3V, and the processor IO pins directly connect to the board IO pins. This can cause relay logic to remain active regardless of whether the output pin is high or low, as current flow goes back through the pin protection diodes on the processor. I discuss this at length in the manual.

If you meet the device specifications for current flow I don't believe operating time is a problem. And the specs are for max environment temperature which you are not likely exposed to.

sjorsBe, 2022-08-03:
So it seems a good idea to put some uln2803 or uln2804.chips on the output pins and limit the current on the inputs 
So more "heavy" loads can be switched without the relay clicking sound. :)
8 ins and 8 outs in one chip. Whenever even more power is needed two ins can be combined in order to have a 1 amp output per input pin. ( but then you need 4 uln's) 
I need them to power a bunch of arduino nanos, each with a number of special tasks to run an elaborate christmas village :)

sjorsBe, 2022-08-03:
The ulns also solve the io pin voltage. The output of the ulns can be a clean 5v

nielsonm236, 2022-08-03:
Well, that approach is certainly a more robust methodology. Depending on your application it may be overkill. I use the off-the-shelf relay modules like those made for Arduino projects - example here: 
https://www.amazon.com/SunFounder-Channel-Shield-Arduino-Raspberry/dp/B00E0NSORY/ref=asc_df_B00E0NSORY/?tag=hyprod-20&linkCode=df0&hvadid=309820150211&hvpos=&hvnetw=g&hvrand=11567779813632940413&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9028722&hvtargid=pla-438786682067&th=1

These work just fine directly connected to a Network Module pin even though they use 5V logic. Per the manual the control pin logic has a LED in series with the opto-isolator that drops enough voltage to prevent excessive current backflow through the STM8 protection diode ... so all works well. Note however that the 16-relay version uses different control pin design (no series LED in the input side ... instead the LED is on the relay side of the logic). As a result the 16-relay version requires care (as described in the manual).

If you are connecting your own bare relays (rather than using one of these relay modules) then you'll need your own interface logic.

I'm not sure what you mean "without the relay clicking sound". A mechanical relay, regardless of the logic, will "click" when it changes state ... and in fact this is an indirect indication that the relay is solidly closing or opening ... which is desirable to minimize contact bounce. A few people have connected solid state relays to avoid the mechanical effects.

If you are getting "relay chatter" (relays that seem to buzz at open or close rather than just a simple "click") you have a different problem, most likely in the power supplies or inadequate current in the control logic.

sjorsBe, 2022-08-06:
I ment that using the uln solution gives power enough to feed the arduino nano's and are silent. So no clicking sound of relays :) 
One uln2803 is sold for 25 eurocent ( 10 for 2,47 with free shipping ) and takes care of 8 outputs. A quarter well spent... :)

nielsonm236, 2022-08-24:
I'm closing this one since the spec provided satisfies the original request for information. Feel free to open a new issue if there are more questions.
Mike

closed 2022-08-24 by nielsonm236