nielsonm236/NetMod-ServerApp #106: 16 channel relay module vs network module pin out

rtmad21, 2022-08-11:
**Question:** How difficult would it be to change the pin order, i.e pin 1 (firmware) triggers pin 2 (hardware)  **_or_** reprogram the REST commands so that 00 turns off pin 2 not pin 1?
  
**Background:** I am working on a project that uses REST commands to trigger the relays on 15 boards (240 total relays) the Relay Board pinout does not match the Network Module pinout. With so many boards and comands simplicity is key.  For now, my 2 options are to either use an illogical REST comand structure for all relays or, individually wire the connection beteween boards (individual wires or a custom ribbon).

Thoughts or ideas?

nielsonm236, 2022-08-12:
My first reaction was "You gotta be kidding ... I wonder how many variations of mapping there are on these 16 channel boards". But, in reality, probably not a lot of variation.
Believe it or not, the firmware logic to control the relays ends up being more complicated that you would think, given the outputs can be controlled from the Browser, MQTT, or REST - all in parallel. And any individual pin could be an input, output, or special function (like temp sensors, I2C EEPROM, and so on).
Having said that, the easiest place to do a remapping would be in the logic that actually writes the pin hardware. So, the next question is "Should this be done as unique firmware for 16 channel relay controllers?" and be done with it? If we start talking about making in optional in the GUI, and still allowing say 14 relays and 2 inputs, or any of the other many combinations we can all think up then we will sink ... there is very little space for code unless you add an external EEPROM (and there goes the ability to have 16 relays because 2 pins are used for the EEPROM).
So, remapping: Not too hard I think.
Adding to main release code: Very hard I think.
Creating a unique code load: Easy to do, adds more steps in code maintenance.
You could also consider cloning the code and writing your own re-mapper. I could probably even to the function for you and let you take it away for your own builds. Of course then future updates might be problematic for you to merge (unless you are better at Git than I am .... which might be pretty easy as I am not very good at it).

A question for you: Are the mappings simply swapped left to right across the pin rows? Can you provide a map that helps me understand if it is pin1/pin2, pin3/pin4, and so on? Or is it pin1/pin9, pin 2/pin10 and so on? Depending on what you tell me I might have some other ideas.

Your thoughts?

rtmad21, 2022-08-12:
You gave some great information and typed out some thoughts that I had as well.  My initial thought was a standalone firmware just for this project, but I wasn't sure about the complexity of making the change. Because it would likely have to be done for every firmware update (depending on how the pinout change was made, would I make the changes needed to each update or would I update my code with the changes from the update)? 

Here is the pinout comparison. The relay board is odd on one side, even on the other, so for a custom ribbon you simply line up every other wire for the first 8, then the remaining 8 in order.  Excluding the GND and 5v+ obviously.

[image]

At the end of the day, the ultimate question I am facing would be:
What option is simpler and reduces the risk of error the most?

- Make a custom code and deploy it to 15 units and potentially fight with updates
- make 15 custom ribbons
- Code my GUI with an "illogical" pattern
- Simply run it as is and number my wires, and mentally ignore the fact that the relays are not "in order"

Let me provide some further insight on my project, (I anticipate 33% of readers getting excited, 33% thinking I'm crazy and the remaining a combination of both)

The relays are being installed on a floating barge in the middle of a lake and will control the ignition of 500+  professional-grade display fireworks.  Because of this, 2 things must be certain, all of the code from my custom GUI down to the Network Control firmware MUST close the intended relay **_and_** my wiring must stay in place during the unavoidable "vibrations".   Splitting and reworking the ribbon increases the risk of a short and misfire (though I feel the risk is limited).  

To finish explaining the project to appease any curiosity, some of the relays will simply be controlled using a single script with coordinated timing, the other relays are controlled by a custom GUI on kiosk converted cell phones "Launch Devices", one relay at a time. These launched devices are given to our guests to provide the excitement of shooting off fireworks from the safety of the beach.  In the future, we plan to extend this through VPN to allow control of launches remotely while streaming the show for our guests that cannot attend. The good news is, that we have 326 days until the 4th of July.  (:

jmcvieira1, 2022-08-12:
Hello i suggest a custom double side pcb to rearrange the connections to suit your needs, solder this small pcb permanently to the pcb of the network module and solder a ribbon cable connector to it, you wouldn't need to fiddle with the firmware and you wouldn't have any problems with vibrations.

nielsonm236, 2022-08-13:
I have to agree with jmcvieira1. This may sound difficult but is really quite easy (and cheap) if you use a Chinese supplier. For instance I use PCBWay and was able to get 10 boards for only a few dollars. But, of course, you have to design your own board and provide the artwork. Do you have any board design experience or tools? I'm asking because this is another thing I enjoy doing ... and maybe I could help you out with it. Anything I do would be free ... this is only a hobby for me. Your costs would just be ordering the boards and assembling them. If interested we can explore further.

yozik04, 2022-08-13:
I guess you will not expose IP addresses of the units to guests to trigger the relay. You will have a service that would relay commands securely to the units and at least log success/failures. So why you just don't add an additinal layer in code that would remap commands to trigger right pins? That would be the most simple solution in my opinion.

nielsonm236, 2022-08-13:
yozik04: Solving problems in software. Now THERES an idea!

rtmad21, 2022-08-13:
I love the internet sometimes.  Just needed some fresh eyes that are not tunnel-visioned on the project.

**jmcvieira1 & nielsonm236:** I literally thought about a "ribbon converter" board:
 
_Relay > ribbon > Converter board >ribbon > Network Module_  

I could not get past having all of the ribbons everywhere and having to mount the new converter board somewhere, at no point did I think, "Hey dummy, just solder the board straight to the Network module" or just make it a HAT with no ribbon at all...

I have never actually designed a PCB, but have always wanted to, what an easy first project to try! 

Needs some tweaking nothing is measured out or anything, but I made this using CircuitMaker pretty quickly: 
[image]

In addition to fixing my problem, this also eliminates the 5v pins as I am changing out the LM2576 5V to fix the other 16 relay issue. 

Thinking about it, I may design a model that changes the pin order and has the inline diodes to have a more "plug-n-play" solution.  With how cheap the PCBs are I certainly would be able to have extras of both on the shelf for the next guy that needs an easy hardware solution to this problem. 

I never realized how simple PCB design could be, I may have found a new addiction.

**yozik04:** I thought about this also. Most of this is controlled through Python scripts, so it would be rather simple. But, I was hoping to avoid it. I honestly have no real reason behind it, but something about fixing it in my software code just bothered me, again no real project-specific reason. I guess part of it was I wanted to see if there was a more universal, plug-n-play solution.

I am going to try the custom PCB route, not only does this give me an excuse to play with PCB design, but as I mentioned above, for the cost, I will make some extras in case anyone else ever needs it.    

Thank you all for the ideas!

nielsonm236, 2022-08-13:
Well, I'm not only in CO, I'm in Broomfield too. Glad we could help, and I'd love to watch the show next year. Send me a link (or tell me where it is) when the time arrives.
Mike

nielsonm236, 2022-08-24:
I'm closing this one since a solution was found. I'm still interested in seeing your results in the future. Sounds like a fun project!
Mike

closed 2022-08-24 by nielsonm236

peoyli, 2023-01-29:
Further discussed and implemented in pull request #120 https://github.com/nielsonm236/NetMod-ServerApp/pull/120