nielsonm236/NetMod-ServerApp #111: Webpage display tracking confusion when clicking Reboot on Configuration page

nielsonm236, 2022-10-02:
I think this is something users will never notice. Documenting here while I investigate.
By design when the "Reboot" button is clicked it should cause a return to the IOControl page as a result of starting the code over, as if from power-on. Instead the Configuration page is re-displayed. That actually seems like "nicer behavior", but I see in  the internal workings of the code that for several seconds both the IOControl and Configuration pages are active in the Browser traffic processing. This is not good and may result in other issues. I suspect TCP/IP connections are not being closed properly, so the Browser is confused (currently testing with Firefox).

nielsonm236, 2022-10-09:
On examination of the Javascript I see (and now recall) that it is deliberately refreshing the current browser page on reboot. This is deliberate as it is a better user experience. However, there appears to be a slight conflict with the c-code in that the c-code wants to display the IOControl page at reboot. I'm seeing some odd behavior in the webpage processing, but the Javascript wins when the c-code page processing times out (at least I think that is what is happening). I can't just remove the c-code as that code also supports the notion of "which page is displayed first if the user just opens a browser with no specific page reference, like /60, /61, etc". This is a low priority issue as the user isn't even aware of it. But it does confuse things (for me) when debugging other issues.

yozik04, 2022-10-10:
I think javascript should control what will be loaded next. Not c-code.

nielsonm236, 2022-10-10:
The problem first occurs at power-on boot. If after power-on boot someone sends a URL request like "192.168.1.182" with nothing following that, then c-code must decide which page to send. Javascript isn't running yet.

So, to prevent that c-code from running and setting pointers to the "default page" when a Javascript requested reboot occurs I need some means of knowing that the reboot is requested from Javascript. Fortunately power isn't lost, so I could put some flag in RAM to let me know that Javascript made the request. That's what is missing here.

Solving the Chicken-and-Egg problem by definition. :-)

yozik04, 2022-10-10:
After boot you should always send IOControl page. My opinion.

nielsonm236, 2022-10-10:
Yep. That's what I do. And that's the problem. After a Power-On reboot it works just fine. The IOControl the code initializes to send the IOControl page when a request comes in.

But if I am on the Configuration Page and click Reboot the Javascript sends a /91 (reboot) and after 5 seconds sends a request for the Configuration page. So, now I have the IOControl page in queue (because of the reboot), and a parallel request for the Configuration page. So both start processing, but the Browser ignores the IOControl traffic. By watching internal code operation I can see it processing and timing out. Shouldn't be coded that way.

So, I need to find a way to prevent that. I need to make some code changes so that I don't initialize with the IOControl page in queue waiting for a request ... instead wait for the first request, and if it does not specify a page OR it specifies a page /60 only then do I send the IOControl page.

This is pretty basic webpage server stuff ... but that's what the code does now.

nielsonm236, 2022-10-11:
Some of what I said above is nonsense. Going back through the code I find that I send webpages only when a GET occurs ... so I can't be sending BOTH the IOControl and Configuration pages on reboot. Still I see that some kind of unexplained incoming (not outgoing) packet processing is occurring when Reboot is requested by Javascript on the Configuration page, and my initial debug seemed to indicate it was IOControl page related. I will have to dig deeper to figure it out.

yozik04, 2022-10-11:
I think it is ok to ignore the request if the device is busy with an other page.

nielsonm236, 2022-10-16:
Just documenting a few results of my deeper investigation. First of all, I didn't get to the bottom of anything. Just created more questions for myself.
I instrumented the uip.c code with a large number of UART output statements trying to find out where the odd calls to the HttpDCall code were coming from. Despite several hours looking into this I did not find the source of the calls of interest. Clearly I didn't instrument exactly the right thing.
But, other things also showed up that don't make sense: a) There is a lot of activity on Port 1409, most of it rejected as "not my IP" and others are directed at the Network Module IP. Regardless, we should not respond to Port 1409 at all. b) Despite my comment in (a) I think I'm seeing connections formed and then reset from time to time - which should not happen at all. FYI, Port 1409 is assigned in the IANA to "Here License Manager" - after some searches I think this is a defunct product so I'm not sure what is doing the probing on Port 1409.
The uip.c code is a minimized "micro ip" interface. So it doesn't filter everything and I suppose could let some stuff through. Then the application (the webserver) rejects most if not all of that. Despite the unexplained events I'm seeing in the instrumentation at the user level all seems to be working correctly. This will continue to annoy me but for now I will set it aside.

nielsonm236, 2022-12-17:
I'm going to close this as "anomalous behavior" since it does not appear to affect operation, and in fact is not visible to the user.

closed 2022-12-18 by nielsonm236