nielsonm236/NetMod-ServerApp #113: MQTT command rate limitation

michalwoz, 2022-10-06:
I have some issue with I think MQTT command rate limit. By default, you get all 16 relays in "Switch" section in HA dashboard.
There is switch which allow to change all relays state with one click. When I use that switch not every command is handled. All relays were ON. Here is result of trying to switch OFF all relays with single click:

[image]

I think that some buffer gets overflowed. Referring to manual I don't expect module to handle such big message rate. I just want to confirm that this is it close to theoretical limit and hardware is OK. I will also think about some workaround. The first thought is to write simple MQTT "proxy" which will add few milliseconds of delay between messages. Any better ideas?

From manual:
> Overall Command rate:
Processing speed is very limited given the functions implemented, so I imagine it will
be easy to over-run the Network Module with state change requests. The code is
single threaded, so whatever function has been requested must be completed before
the next can be addressed. More testing needs to be done to determine if packets are
simply dropped (if too many received) or if there are cases where the module may
stop functioning. So far, I haven’t seen a “stopped functioning” scenario. If that were
to occur a power cycle may be the only recovery option.

nielsonm236, 2022-10-06:
I suspect you are right - command over-run. At least it didn't crash ... which is what I worked on when I was deliberately over-running the device during testing.
Makes me wonder if an "All ON" and "All OFF" MQTT command would be useful, like what we have in the Browser.

michalwoz, 2022-10-06:
"All ON" and "All OFF" will not be enough in my case.
My tests are not representative, I just clicked 30 or 40 times, but can confirm, that it didn't crash the device. Result is always as on screenshot.
What would be very helpful is finding maximal allowed constant MQTT command rate (which is the same as minimal delay between MQTT commands)

sjorsBe, 2022-10-07:
To be honest, I have the same problem but would be helped with an all of and all on possibility 

greetz

sjors

yozik04, 2022-10-07:
@michalwoz Would be great to see TCP dump to understand what module responds to MQTT messages which are over rate. I am not sure if setting QoS to 1 on home assistant side will help. Maybe something has to be implemented on NetModule side as well.

yozik04, 2022-10-07:
From what I read from the MQTT spec NetModule needs to be coded to handle QOS 1. Otherwise broker shoots messages in fire and forget mode.

nielsonm236, 2022-10-07:
Yes, we are running QOS 0. Code is much simpler (and smaller). We must not forget this device was never intended to be fast. It has a tiny little processor and very limited flash and RAM.
I can go back and look at QOS 1 again, but first I have to figure out why expanding the uip_buf for Issue #109 is causing a bunch of problems.

nielsonm236, 2022-10-08:
Take a look at Issue #112. If implemented in REST and MQTT that might solve the problem for "ALL ON/OFF" and perhaps @michalwoz issue. In MQTT (and REST) there would be a single message that then switches all the outputs according to a setting bytes. The firmware would filter so that Output pins are the only pins affected, and a user settable mask could also be implemented so that only specific output pins are affected by the command. This would eliminate a bunch of messaging and message processing.
Mike

yozik04, 2022-10-10:
This won't fly as HomeAssistant does let you to override that toggle all switch as I know. It will just toggle all which will initiate a burst of MQTT messages to NetModule

nielsonm236, 2022-10-10:
Hmmmm. Not sure I understand. If we create an MQTT command to send a set of on/off states in a pair of bytes, that message is sent to the Network Module, the Network module switches the outputs and sends a series of messages to HomeAssistant with the state of each individual output. Agreed that HomeAssistant has various MQTT level interactions as each Output state is reported by the Network Module, but those state change messages are effectively throttled by the Network Modules ability to send the Output state messages to HA. So I don't think an overload occurs if done this way. This is effectively what happens if a user uses a Browser to change output states ... or the user uses the /55 (All ON) or /56 (All OFF) URL commands.

Having said all that ... I guess I'm not sure how we create a user interface in HA that let's us send a such a command. Must be a way. I certainly know how to define it at the MQTT level.

yozik04, 2022-10-10:
[image]

See that toggle in the header. If it is toggled it will send status changes to all of the below. It is not possible to control it's action and override it to send just one request. So it is not a fool proof approach.

nielsonm236, 2022-10-10:
Interesting. I always have that disabled in my HA.
Why would it be useful to send status back to the Network Module? The Network Module doesn't do anything with it. Just more MQTT traffic that is discarded.
Tonight I will experiment with the current version, using the /55 and /56 commands to send a lot of traffic to HA to see if the returned status messages have any effect. Even though I might be discarding them they still fill message slots and the incoming queue. I never saw a problem before but admit I didn't do a lot of testing on it.

nielsonm236, 2022-10-10:
I just tried simply flipping the toggle in the header left and right WITHOUT any deliberate Output changes and got all sorts of bad behavior. Outputs changing state and so on. What exactly is that toggle doing?

nielsonm236, 2022-10-10:
With a little reading I think I understand. And re-reading your message I believe I misunderstood it. By "If it is toggled it will send status changes to all of the below" you must have meant that it sends a state change message to each of the entities to toggle their state from on-to-off or off-to-on. That is exactly the over-run problem that @michalwoz submitted but he used a different method to generate the MQTT messages. So, we already know that we can over-run the Network Module receive queue with a lot of incoming messages, and I don't see any way to fix that. This being the case the Toggle at the top of the HA card is useless with the Network Module. Assuming, of course, I understand your point and what the Toggle does.

So, back to our discussion. If we know you CANNOT rapidly send 16 MQTT commands to the Network Module to change 16 outputs (and perhaps can't even do it for 3 or 4 outputs) - what other option do we have? And we are back to the idea of using a Mask and single command with Output state settings. The key is "single command". That get's through to the Network Module just fine, it changes the states, and transmits up to 16 state change messages via MQTT (if MQTT is enabled).

I just tried the REST /55 and /56 commands and had no trouble. HA is reporting the state changes just fine. This is why I believe this method will work.

The HA card Toggle doesn't work for multiple outputs. Never has as far as I know. But I never used it so was unaware that it would over-run the receive buffer.

michalwoz, 2022-10-10:
I will take a look on QoS, but as you wrote this may not be possible to implement QoS due to memory limits. I don't even know yet if that will solve such kind of problem. Will check.

In my case I can accept constantly running rate limiting service, that will accept any message rate from HA or any other source and which will send those messages to Network Module but not often than XX milliseconds.

nielsonm236, 2022-10-10:
@yozik04 I just tried an experiment with HA Toggle. The Network Module worked with Toggle with up to 3 outputs configured. At the 4th one it started having trouble.

@michalwoz I think I will still pursue the MQTT/REST "batch" message approach, as @sjorsBe is interested, and there was a request long ago for something similar. I don't know how that could be implemented in HA (I'm sure there is a way), but many other applications like NodeRed could easily use it. And it could end up also allowing an All-ON/All-OFF button on the IOControl page.

nielsonm236, 2022-10-11:
@yozik04 Just FYI (you probably already know this) - In HA I "Edit Dashboard", then I Edit each card. When I open Edit for the card I get the following which includes the option to turn off "Show Header Toggle". That's the way I've always used it, with toggle off. Unfortunately displaying the header toggle is the default. I will probably have to include this in the manual.
[image]

yozik04, 2022-10-11:
QoS 1 would be the best solution.

nielsonm236, 2022-10-11:
OK. I will bump up the priority on this since it is sounding more like a defect rather than an enhancement. I'll need to compare the original MQTT code to what we have now to seek any deletions or modifications related to QOS1. Not sure of the code space needed. I don't think it adds much if anything to RAM usage. I'll create a branch for this as it has the potential to introduce some bugs. This won't improve performance, but also shouldn't hurt it much. But if it works reliability will be much greater across the MQTT management products.

sjorsBe, 2022-10-11:
You see, we keep you busy until you are 95 years old:)

nielsonm236, 2022-11-06:
This issue has been very difficult to resolve. I made a few changes to code to regulate message queuing a bit better, fixed one bug in the Network Module MQTT PUBLISH Response process, but also found that Mosquitto version 1.6.x must have had some bugs which were resolved at some point ... when I upgraded to Mosquitto 20.0.15 a lot of problems went away. I'm going to make code available in a Beta context soon so others can try it out. The one problem I have not been able to resolve is that when a Network Module reboots for some reason the Toggle function in Home Assistant will not toggle all pins on/off until the Toggle function has been executed 3 to 5 times. Then it starts to work properly. I can see from Mosquitto logs that HA is not issuing all the PUBLISH State Change messages until that 3 to 5 attempts has been made, then everything is in sync and it seems to work properly from then on ... until the Network Module is rebooted again. I'll be interested in hearing what others find once I get the Beta released.

nielsonm236, 2022-11-09:
Look for Pre-release 20221109 0109. It has the fixes for the Home Assistant "Toggle" function. The pre-release is for test only.

nielsonm236, 2022-11-13:
See issue #122. QOS 1 may not be of any value here. The code changes to send a PUBLISH Response regardless of whether a received PUBLISH State Change actually changed the state may be the best answer, and I've implemented both.

nielsonm236, 2022-11-19:
Per issue #122 I'm likely to abandon QOS 1 due to the MQTT spec requirement that there be a disconnect/reconnect. Clearly the spec is trying to use QOS > 0 only for network disruptions that then require resending messages. But that's not what we are seeing here.
As an update, I've turned my attention back to QOS 0 operation. I'm beating my head against a wall trying to figure out what is happening here. I've got enough instrumentation applied to see that Home Assistant is sending all the commands, and those commands are being sent to the Network Module (curiously that wasn't the case with QOS 1). However, I haven't been able to determine why the commands are not all getting processed. In fact, the MQTT code doesn't even see all of them ... and I'm not getting errors from the ENC28J60 hardware ... so my code (or the Liam Bindle uip.c code) has a bug somewhere in the disposition of incoming messages. Every one of the switch toggles work individually, but not as a batch. Back to the wall ...

nielsonm236, 2022-11-20:
Well, managed to beat my head part way through the wall.
Short version of a long story: When an MQTT source wants to send multiple MQTT PUBLISH messages to the same target it is allowed to "batch" them into a single TCP packet. I didn't design for this ... my current implementation expects one MQTT PUBLISH per TCP packet. I will work on a fix.

nielsonm236, 2022-11-25:
Update: I have a fix. Doing my own testing right now, but so far it is good.
Changes: 
Had to rearrange RAM usage to allow a large uip_buf during initialization (primarily for HA Config message length), then reuse some of the space as an MQTT receive buffer after HA Config messages are transmitted.
Implemented reconstruction of incoming MQTT messages that have been batched into one TCP datagram, including bridging of the packets when the datagram crosses TCP packet boundaries (that was a PITA).
Deprecated the feature allowing addition of a DS18B20 sensor "hot". Going forward adding a physical sensor requires a reboot so that the associated HA Config message is generated only during boot (not during runtime). This is to accommodate the multi-use of the uip_buf space.
Implemented separate outgoing and incoming MSS values for TCP connections. Original code selected the smallest advertised value and used it for both directions.
There is still some debris hanging around that needs to be cleaned up: a) Can simplify the DS18B20 code a little due to the deprecation of hot-add, b) Found a couple of "generic" code spots in uip.c and mqtt.c that can be deleted, c) Need to remove all QOS 1 code (currently enclosed in if/endif statements so it doesn't affect code size), d) Need to review some use of array[index] statements where pointers might be more efficient, e) Need to review the MQTT message and packet reconstruction code for efficiency.
I will try to put up a pre-release once I am convinced it is adequately stable.

nielsonm236, 2022-11-29:
Addressed in pre-release 20221127 0000

nielsonm236, 2022-12-04:
If you've tried this and it appears to work for you please leave a comment, even if only "Looks OK".

nielsonm236, 2022-12-17:
Addressed in release 20221217 1925.

closed 2022-12-17 by nielsonm236

michalwoz, 2023-01-15:
@nielsonm236 I also have tested that. It works too. Thank you.

I don't have any issues with the order or received "responses". They looks like that:
```
20230115-17:50:55.226638 NetworkModule/RelayA/output/01/set OFF
20230115-17:50:55.266760 NetworkModule/RelayA/output/02/set OFF
20230115-17:50:55.267501 NetworkModule/RelayA/output/03/set OFF
20230115-17:50:55.267706 NetworkModule/RelayA/output/04/set OFF
20230115-17:50:55.267837 NetworkModule/RelayA/output/05/set OFF
20230115-17:50:55.267986 NetworkModule/RelayA/output/06/set OFF
20230115-17:50:55.268443 NetworkModule/RelayA/output/07/set OFF
20230115-17:50:55.268617 NetworkModule/RelayA/output/08/set OFF
20230115-17:50:55.269029 NetworkModule/RelayA/output/09/set OFF
20230115-17:50:55.269524 NetworkModule/RelayA/output/10/set OFF
20230115-17:50:55.269808 NetworkModule/RelayA/output/11/set OFF
20230115-17:50:55.269962 NetworkModule/RelayA/output/12/set OFF
20230115-17:50:55.270098 NetworkModule/RelayA/output/13/set OFF
20230115-17:50:55.270227 NetworkModule/RelayA/output/14/set OFF
20230115-17:50:55.285005 NetworkModule/RelayA/output/14 OFF
20230115-17:50:55.336420 NetworkModule/RelayA/output/13 OFF
20230115-17:50:55.387258 NetworkModule/RelayA/output/12 OFF
20230115-17:50:55.438495 NetworkModule/RelayA/output/11 OFF
20230115-17:50:55.489847 NetworkModule/RelayA/output/10 OFF
20230115-17:50:55.540671 NetworkModule/RelayA/output/09 OFF
20230115-17:50:55.591901 NetworkModule/RelayA/output/08 OFF
20230115-17:50:55.643859 NetworkModule/RelayA/output/07 OFF
20230115-17:50:55.695032 NetworkModule/RelayA/output/06 OFF
20230115-17:50:55.745934 NetworkModule/RelayA/output/05 OFF
20230115-17:50:55.796844 NetworkModule/RelayA/output/04 OFF
20230115-17:50:55.848108 NetworkModule/RelayA/output/03 OFF
20230115-17:50:55.900036 NetworkModule/RelayA/output/02 OFF
20230115-17:50:55.951262 NetworkModule/RelayA/output/01 OFF
```

Sometimes first response comes just after first "SET":
```
NetworkModule/RelayA/output/01/set OFF
NetworkModule/RelayA/output/01 OFF
NetworkModule/RelayA/output/02/set OFF
NetworkModule/RelayA/output/03/set OFF
```

It is just my observation, informational, it's OK for me. Thanks again.

nielsonm236, 2023-01-15:
In my testing I noticed that the first one usually came in "alone", thus getting a response from the Network Module before the batch of messages from Home Assistant. I don't know why that happens ... probably some timing issue within Home Assistant.
Thank you for the feedback!