nielsonm236/NetMod-ServerApp #114: Technical diierences between the 8 port and 16 port module

sjorsBe, 2022-10-07:
I just ordered a few 8 port modules. 
The entire hardware seems to be the same. Is the difference just software or are there hardware limitations on the device itself?

kr 
Sjors

nielsonm236, 2022-10-07:
It is just a firmware limitation. I did not dig deep enough to figure out how the factory firmware knows it is on an 8 port board or a 16 port board. I think there is a resistor difference feeding an analog input. But my firmware does not check those inputs, so every board operates as 16 ports.

sjorsBe, 2022-10-08:
Hi Michael, 

thanks for your answer. So its wise to buy just the 8 ports model and save a few euro. 
Im very pleased with the modules mainly thanks to your amazing work!
Ive combined them with homebridge on a raspberry pi 8 outputs with 8 inputs to “locally” switch the outputs.
It now all works through mqtt and pi, so i do understand the question i read to be able to really link  a local input 
Pin to an output pin ……
So yet another question for functionality to be squeezed in the little space you have. :) 

kind regards 

sjors

nielsonm236, 2022-10-08:
Might have to open yet another Issue for the linking an input pin to an output pin. I'm not sure what is being asked. Is this as simple as adding a wire from the input pin to the output pin, but you want to make the connection in firmware?

sjorsBe, 2022-10-08:
What i mean is the following. 
Using one module and configure it with 8 inputs and 8 outputs
Changing input 1 should activate output 1
Changing input 2 should activate output 2 
Etc. 

In this way the midule can be used with standard switches and with smarthome devices 

kr 
Sjors

nielsonm236, 2022-10-08:
I opened Issue #116 for the "link input to output" discussion.

nielsonm236, 2022-11-06:
I'm closing this issue as I think I answered the initial question.

closed 2022-11-06 by nielsonm236