nielsonm236/NetMod-ServerApp #116: Link Input pin to an Output pin

nielsonm236, 2022-10-08:
sjorsBe requested the following:
Would like to map a change on and input pin to cause action on an output pin. For example:
Using one module configured with 8 inputs and 8 outputs
Changing input 1 should activate output 1
Changing input 2 should activate output 2
Etc.
In this way the module can be used with standard switches and with smarthome devices

My initial comments:
Seems I discussed this with another user at some time, but I can't find that discussion in Issues. Might have been by email or in Youtube comments. At the time I think it was decided this functionality should be performed at a higher level, perhaps in HomeAssistant or similar control software. Since this is coming up again I'm willing to reconsider, and now we have an Issue to discuss and record conclusions.
I recall the original request was thought to be a bit complicated as a GUI was required to map the selected input pin to an output pin. But this could be simplified greatly if it is ALWAYS input1 maps to output1, input2 maps to output2, etc. If that is the assumption then an additional pin Type (perhaps "Mapped"?) could be defined to tell the firmware that the pin is an input and it is mapped to a predetermined output.
The underlying complications:
- "Mapped" would only be allowed on pins 1 to 8
- The corresponding pin must already be defined as an Output
- If the corresponding pin is defined as Input, Disabled, or is already used as a special function (I2C or Temp Sensor), then selecting "Mapped" will cause the pin to be forced back to Disabled. The user will need to be smart enough to realize that "Mapped" is disallowed on the pin as providing online feedback would be complicated. At least I think it would be complicated.
- The user would first have to define the Output pin as Output, SAVE the change, then define the Input pin as "Mapped".

I'm not sure what complications will be encountered. As it stands the Input change will cause a fairly fast Output change (milliseconds ... but could be hundreds of milliseconds depending on what else is happening). That time probably cannot be well defined.

This would probably be a feature for all release types (non-updatable AND updatable). It will depend on how much code is involved.

So, another idea on the queue.

Mike

sjorsBe, 2022-10-08:
All,

it would be great to have some kind of solution like that.
It brings the possibility to automate e.g. standard wall switches to be used with a netmodule and relais . 
it brings the possibility to be able to switch locally AND as smart device….

i spent some time in the Sonoff world, where people liked to make that happen with sonoff basics. And ended up with extremely nasty and risky solutions. This could be a clean and stable solution…..
Please give your thoughts 

kr 
Sjors

yozik04, 2022-10-10:
With standard wall switches there will be a more complex solution.

Let's assume if we map input to output.

1. Input is wired to the switch and circuit is OPEN, Output is OFF
2. Ok, we toggle the switch state becomes the following
3. Input circuit is CLOSED, Output is ON, all fine.
4. Now we toggle input through the network.
5. Input circuit is still CLOSED, but Output should be OFF. Hmmm. Mismatch..
6. Let's toggle switch now.
7. Input circuit is now OPEN, but Output should be ON.... Not the way it was in the beginning.
8. Let's toggle again through the network
9. Input circuit is OPEN, Output is Off, all has returned to the initial state.

So the NetModule would need to hold a virtual switch (network) state somewhere and it should survive reboot. Otherwise you would have some random lights ON after every blackout. I am sure you will not like that.

yozik04, 2022-10-10:
To work with switches
1. Output needs to be in initial state (defined in UI) after every reboot. Regardless of Input.
2. Input state change should toggle output. Regardless of if it was OPEN or CLOSED.
3. Network command should set output state to what was sent in the packet.

jmcvieira1, 2022-10-10:
Hello
If we use a push button stile switch the input circuit is closed for a moment,(doorbell button) but the network module needed to toggle de output on or off at each button press.

sjorsBe, 2022-10-10:
cant we use the "invert" setting of the module to tackle the problem? 
whenever the network toggles the invert setting,  it looks like the entire issue is solved Or am i not overseeing the whole issue?. Add  payload "invert" to mqtt including the client software behind it. 
 
kr 

Sjors

yozik04, 2022-10-10:
Invert is a bit in memory stored on chip. You do not want to toggle that.  Chip has a finite amount of writes.

sjorsBe, 2022-10-10:
It should be possible to load the invert state to ram while booting the netmodule (max 16 bits) and redirect the status invert requests to that position in ram. 
Whenever the netmodule  boots it looks at the bootstate setting which is a separate setting. And updates the ram invert setting to functional off depending the linked input. 

So 
If switch is in on position while booting ...
Netmodule looks at the flash setting.
If it is not set, it looks at the input setting whenever that setting is set it sets the ram invert bit to "on" in order to obtain a functional off. So just on the bootcycle it needs an extra check when an input output link is in place. 

I hope i explain it so it is understandable :) 
I think that should work .. and wont consume much ram and save the flash writecycles

yozik04, 2022-10-10:
Question if we have enough RAM for that =) I understood we have some code space left but almost no RAM. @nielsonm236

sjorsBe, 2022-10-10:
Its a lousy 2 bytes :)

nielsonm236, 2022-10-10:
The two bytes I'm not so worried about (although we are working with the last 20 or so bytes of RAM ... and I still want to get some promised sensor work done for @jmcvieira1). Maintaining current functionality plus implementing this functionality is of greater concern. I need to think through what @sjorsBe said above.

By the way, the "Mapped" setting I described above is stored in EEPROM. So that cannot be changed frequently (as @yozik04 described there is limited writes to the EEPROM).

@sjorsBe Above you say "Netmodule looks at the flash setting." Which setting?
Above you say "If it is not set, it looks at the input setting ..." Which setting?
Just looking for clarity so I'm not misunderstanding.

I think we need a small boolean table to make sure this will always work through boot (or multiple boots).

nielsonm236, 2022-11-11:
A concern with the suggestion that the Invert settings be loaded to RAM and operate from there. If power is lost (or glitched) the Invert settings in RAM will be lost, and it might be very confusing to the user to get things back in sync. That is why we store them in EEPROM (to survive a power glitch or reboot). So this needs some more thought.

Another thought: Since this function is directed at making an Input pin control an Output pin, when a pair of pins are "mapped" I could disable MQTT, Browser, and REST from changing the Output state. Only the Input pin could change the Output state.

What is the use case where pins are mapped, but we also want MQTT/Browser/REST to change the Output state? I suppose it could be the case that there is some remote light that was turned on with a remote physical switch (say out at a storage shed), then later you want to turn the light off from the house. If that is the case I like the @jmcvieira1 suggestion that the remote physical switch should have the properties of a "push button". It creates a pulse that toggles the state of the Output. Then the MQTT/Browser/REST command can see the Output state and can toggle it again if needed later. Any switch can generate a pulse with a minor bit of hardware added. And having said that ... I could perhaps write code that would detect a switch edge rather than the switch state, and use that edge as the push-button detect. Of course this requires software debounce. That way any time the switch position changes I can flip the Output state. But that switch now has the properties of a single-pole-double-throw switch ... it doesn't have an "ON" state, only a "change it from what it is now" function. Does that work?

sjorsBe, 2022-11-14:
Hi Mike

So the module should keep on supporting the mqtt protocol to be connected to the smart actuators in the house. 
The local switch option is a failsafe which can be easily explained to elderly people.

nielsonm236, 2022-11-14:
I like the described use case - good idea. And I understand the elderly people issue ... that's my resting state.

Does my suggestion of flipping the output state if I see an edge on the input work? If the physical switch is moved the output state is simply flipped. So, it could be a toggle switch or a push-button - either will function the same way.

nielsonm236, 2022-12-01:
@sjorsBe I'm looking at this code now while I wait on feedback from the latest pre-release.
As I was looking at code your comment "keep using my wall switches" kept coming back to me. I'm not sure what this means - do the wall switches remain connected to the load as they were originally? How is the Network Module wired in to also control the load? I've attached a drawing showing what I think you have in mind. Let me know if I have this right. Obviously not all detail is present in the drawing. Just the concept.
Mike
[image]

sjorsBe, 2022-12-02:
Hi Mike,

You are right with the second drawing.
As you know, I live in europe, the majority of wall switches are SPST. Or simply: the standard “ON / OFF switch.

The idea is to let the wallswitch switch the netmodule ( on /off ) which in case uses a solid state relay to switch the power load. ( eg a lamp)
The module should be able to switch “locally” when there is no tcp/ip network available.
Whenever a network is available, the module should be switched “locally” or over the network.

Something to be considered:
Whenever the netmodule powers up, the solid state relay should be in the position as was defined in the setup of the netmodule, as was defined in the “boot state” setting. Regardless of the setting of the SPST wall switches.
(This means that the setup of the netmodule should safeguard a “safe state” of the solid state relays. )

I hope to have shed some light in the topic, but feel free to let me clarify the functional specifications ☺

Kind regards

Sjors

Verstuurd via prive mail.

[SjorsBe Dwg1]<https://user-images.githubusercontent.com/63531544/204951187-94f233a5-e79f-430e-b224-7de7b6d2021c.PNG>

yozik04, 2022-12-02:
> Hi Mike, You are right with the second drawing. As you know, I live in europe, the majority of wall switches are SPST. Or simply: the standard “ON / OFF switch. The idea is to let the wallswitch [...]

I use zigbee modules the same way. I install them behind the switch. Works great.

nielsonm236, 2022-12-02:
OK, good. So now let's go a little deeper into the logic of how this should work. Referring to your elderly example, I can imagine that they would like the light on when the switch says "ON", and the light off when the switch says "OFF". It might be confusing to them to ignore the "ON/OFF" markings on the switch. But perhaps not really a problem?

Two approaches (A and B):
A) If I am powering up and the Boot State is "OFF" then I will turn the light OFF regardless of the switch position. If the switch is in the ON position I suspect most people (even the elderly like me) would find that curious and would instinctively turn the switch OFF then back ON again. The person operating the switch would likely think "Oh, now it's working" and walk away. I can ignore the switch going from ON to OFF since the light is already OFF, and when I see the switch go from OFF to ON I can turn the light ON. The same scenario would work in reverse if the Boot State was "ON".
A practical example: Grandfather is sitting in the living room with the light ON. There is a brief power outage and the light goes OFF. He walks to the switch and finds that it is ON. So, he'll likely turn it OFF, then ON, and the light comes back on.

Is the above what you want? 
OR

B) Would you want to follow my earlier suggestion that we ignore the fact that the switch has an ON/OFF marking and simply flip the state of the Output if the switch changes position? Unfortunately this would mean that sometimes the ON position is OFF, and the OFF position is ON, depending on the position of the switch when the Network Module boots. I now don't think I like this as much because:
Since booting should be infrequent, it is possible that months go by with ON meaning ON, then there is a power glitch and now ON means OFF.
The same practical example: Grandfather is sitting in the living room with the light ON. He has grown accustomed to the light being ON when the switch is ON. There is a brief power outage and the light goes off. He walks to the switch and finds that it is ON. When he turns the switch OFF, the light comes back ON. When he turns the switch ON, the light goes off. Then he starts looking for ghosts.

I personally like the (A) approach primarily because of the markings on the switch matching the state of the light except for the first step after a power outage. So let me expand on (A):
If the switch is ON and the light is ON, then MQTT turns the light OFF, the user would have to go through the same steps of turning the switch OFF then ON to turn the light back ON.

Your thoughts?

Mike

nielsonm236, 2022-12-02:
@yozik04 Hi Jevgeni - Yes, as I walk through this I was thinking that something like zigbee is also a good solution. It is WiFi based (right?) so the only difference is that the Network Module is hard-wire based, which some people prefer (to avoid WiFi interference, or whatever their reason). So, I don't mind implementing it if it is a small Flash footprint.

nielsonm236, 2022-12-02:
@sjorsBe I had to do a little checking ... in the US SPST switches are actually marked with ON and OFF, with UP typically being ON, and DOWN typically being OFF. In Europe it looks like marking ON/OFF might not be very common (if at all), but the position of the switch is customarily UP for OFF, and DOWN for ON (I think). But my logic above remains the same, just adjusted for local code and custom. :-)

sjorsBe, 2022-12-02:
In europe we dint have “on” or “off” on wallswitches. :)

Verstuurd via prive mail.

nielsonm236, 2022-12-03:
OK - but you do have the notion of "press the top for off" and "press the bottom for on", correct? If yes, this is the on/off pattern that would get reversed with method B. So we need to decide if we want method A or method B.

nielsonm236, 2022-12-04:
OK. So if one day toggling it one direction turns the light ON, but the next day toggling it the other direction turns it ON, that would be OK?

yozik04, 2022-12-04:
This is how it works with my zigbee modules now.
Zigbee is Wireless but now WiFi =)

nielsonm236, 2022-12-04:
Based on this I'll go with Method B and see if anyone complains. :-)

nielsonm236, 2022-12-17:
Addressed in release 20221217 1925. See the manual for operating notes.

closed 2022-12-17 by nielsonm236