nielsonm236/NetMod-ServerApp #118: Reasons not to respond MQTT msg to every set msg, but only on change

michalwoz, 2022-10-11:
What are the reasons not to send MQTT message to topic NetworkModule/NAME/output/XX every time when MQTT message is sent to NetworkModule/NAME/output/XX/set?

Now MQTT message to topic NetworkModule/NAME/output/XX is sent **only when output changes** its state. When we send set message with the same payload in which state output is already in we don't get any "response". I ask this question, because after few days of running lights, I hit the inconsistence between NetworkModule/NAME/output/XX and NetworkModule/NAME/output/XX/set states. As a result, I couldn't turn off the light, the switch in HA comes back to ON position every time after few seconds. Sending message every time will solve this issue. I think that it may generate many other issues.
[image]

nielsonm236, 2022-10-11:
The way the code determines which output state change messages to send over MQTT is by comparing the "before" and "after" states of the outputs. So, if a duplicate "set" instruction is sent by MQTT no change occurs, before = after, and no message is returned to the broker. It is possible that more than one output has changed state before the changes are detected, so at that point the only way I can tell that a state change message was received is to compare the "before" and "after" state registers. Keep in mind that in parallel with the MQTT messages someone could change a relay state via the Browser ... so again the before/after registers enable the code to tell the broker that a state change occurred independent of the MQTT messaging.

If a "set" message was sent to the Network Module and the "set" didn't actually occur, then we have some other problem like a dropped MQTT message.

If we get QOS 1 working does this solve the problem? If a state change actually occurred an ack message should always be returned to the broker.

nielsonm236, 2022-10-11:
What is the time scale in the chart you sent?  Are there any other outputs being changed at the same time? Could more that 2 or 3 output change requests have occurred almost simultaneously?

nielsonm236, 2022-11-06:
I was seeing this in my development of the QOS 1 code. After updating Mosquitto to version 20.0.15 I have been unable to reproduce the problem. Of course there have been other Network Module code changes during this development, but the problem you describe above persisted until the Mosquitto version update. What I was seeing in the Mosquitto logs is that the Network Module would PUBLISH the change in the pin state, but Mosquitto did not forward the message to HA. What version o Mosquitto are you running?

nielsonm236, 2022-11-06:
Additional comment: Google search on "home assistant card switch states out of sync with app" showed a number of people having the problem that Home Assistant and the actual state of a switch getting out of sync. The Home Assistant replies all seem to indicate that the state reply was not received by Home Assistant. OK - I figured that out on my own. The real question is "why not?" So this is probably a fairly common problem. It has reduced to "no longer happening" on my test bench by updating Mosquitto to 2.0.15, so that may have been my primary cause of the issue. But I find this to be an unsatisfying answer as it seems over a very long time other messages could be lost due to a) Excessive activity in the Network Module, b) Another Mosquitto bug, c) A Home Assistant bug, d) A network interruption, d) etc.  So some thought needs to go into how to assure that Home Assistant is not out of sync. QOS1 could be implemented for PUBLISH Response messages as I have currently done for the PUBLISH State Change messages, but I'm not sure how to implement and test all this to be sure it REALLY works given the limited resources of the Network Module and my test capabilities. I wonder if something dumber and simpler could be used like simply resending the state of all output pins at some interval. But again with limited Network Module resources we've got to be sure we don't over-whelm the Network Module with a bunch of traffic (either with more QOS1 or more pin state messages).

nielsonm236, 2022-11-09:
Look for Pre-release 20221109 0109. It has the fixes for this - available for test now.

nielsonm236, 2022-11-29:
Addressed in pre-release 20221127 0000

nielsonm236, 2022-12-04:
If you've tried this and it appears to work for you please leave a comment, even if only "Looks OK".

nielsonm236, 2022-12-17:
Addressed in release 20221217 1925.

closed 2022-12-17 by nielsonm236

michalwoz, 2023-01-15:
@nielsonm236 I've just updated two Network Modules. It works! Good job, thank you very much.
```
NetworkModule/RelayA/output/02/set OFF
NetworkModule/RelayA/output/02 OFF
NetworkModule/RelayA/output/02/set OFF
NetworkModule/RelayA/output/02 OFF
NetworkModule/RelayA/output/02/set OFF
NetworkModule/RelayA/output/02 OFF
NetworkModule/RelayA/output/02/set ON
NetworkModule/RelayA/output/02 ON
NetworkModule/RelayA/output/02/set ON
NetworkModule/RelayA/output/02 ON
NetworkModule/RelayA/output/02/set ON
NetworkModule/RelayA/output/02 ON
```

nielsonm236, 2023-01-15:
Good news! Glad that worked. This and #113 were difficult.