nielsonm236/NetMod-ServerApp #119: Full duplex on ethernet interface

michalwoz, 2022-10-25:
What is Full Duplex checkbox responsible for?
It looks that even if it is checked Network Module advertises only 10M half duplex.
When forced to 10M full duplex it works with full duplex.

[image]

nielsonm236, 2022-10-25:
That is described in the "Notes on Feature Settings" section of the manual. I recommend that you DO NOT use it unless you are using Cisco Business switches. Text from the manual:

The Full Duplex checkbox determines the Half / Full Duplex Ethernet communication method. The default setting is Half Duplex because that is the most reliable setting for the ENC28J60 Ethernet chip. Since all the Ethernet transactions that will occur with the module are small and infrequent there is no real performance advantage to using Full Duplex.

During test it was found that Cisco business level switches exhibited Half Duplex timing that the ENC28J60 cannot handle, the symptom being a device disconnect (and automatic recovery) every few hours. This isn’t the fault of the Cisco switch, rather it appears to be the fault of errata in the ENC28J60. During test it was determined that we could get around this issue when connected to the Cisco switch by enabling Full Duplex mode in the ENC28J60. While there was concern that this would not work (due to chip spec notes and online discussion over the years), it seemed to run error free. There may be other switches which show the same issue. Again, the reason Full Duplex works may be the very low messaging rate used with the Network Module which eliminates the need for flow control.

Note 1: The spec for the chip indicates that Full/Half Duplex auto-negotiation DOES NOT work. However, experimentation showed that both Full and Half Duplex worked with some unmanaged switches, but not with others. Problems were always running Half Duplex only with the Cisco 1G managed switch. No problem was seen running Half Duplex with a Cisco 10/100 managed switch.

Note 2: If you choose to use the Full Duplex setting note that the spec says the Switch port the device is connected to MUST be manually configured for Full Duplex operation … even though our testing did not always show that to be the case. Of course we had a limited number of switches and this might be an issue on some other switch.

Note3: Feel free to experiment with this setting at your own risk to see what works best in your network configuration. I recommend you use Half Duplex and only try Full Duplex if you have issues.

nielsonm236, 2022-11-06:
I'm closing this issue as I think I answered the question.

closed 2022-11-06 by nielsonm236