nielsonm236/NetMod-ServerApp #122: Need to verify that QOS 1 is actually working

nielsonm236, 2022-11-07:
Related to fixes applied for issue #118 and #113 (added QOS 1 for PUBLISH State Change messaging from HA): During test I became suspicious that loss of a QOS 1 PUBLISH message sent from the Broker to the Network Module did not result in a resend of the PUBLISH. I need to come up with a way to test this that can easily be detected with the tools I have at hand.

Thought: add some debug code that will deliberately omit a PUBACK at an interval I can easily see in the Mosquitto log. I should then see a resend of the PUBLISH. Maybe configure a Network Module with a single output, then drop the PUBACK on every 5th button press in HA. Halt all other MQTT traffic on other test modules to simplify the log.

Someday I need to learn Wireshark.

nielsonm236, 2022-11-13:
Well ... implementing QOS 1 appears to have been a major waste of time ... other than learning a bunch of stuff.

I just implemented more debug code to allow me to skip sending a PUBACK on receipt of a QOS 1 PUBLISH from the Broker. My code is reporting that no message resend was received from the Broker, and on looking at Mosquitto logs I see that no resend was performed. Now that I knew more about what to search on to find an answer, I find this:
https://stackoverflow.com/questions/67589558/mqtt-3-1-1-broker-qos-1-at-least-once-message-redelivery

The summary: The only circumstance where a Client or Server is REQUIRED to redeliver messages is on a reconnect.
What???? So, what good is QOS 1? How often should I reconnect to make sure messages are being received???? This sounds useless.

Given this new knowledge I may have to back out the QOS 1 code to save Flash space.

Interested in feedback from anyone that knows MQTT better than me.
Mike

nielsonm236, 2022-11-16:
After some interaction with Steve on "Steve's Internet Guide" http://www.steves-internet-guide.com/connected-or-disconnect/?unapproved=318503&moderation-hash=ddd2837dade6c5fde666ed4b7b243690#comment-318503 I've come to the conclusion that if we want QOS 1 to work properly I will need to add frequent disconnect/reconnect messages. And by "frequent" I mean with almost every PUBLISH from the Network Module.
Concern: This adds a lot of MQTT traffic. Now, the MQTT traffic on Ethernet is not so much a problem, but because of limited RAM there really isn't room for large queues, so my solution on the Network Module is to manage all queues with state machines. Unfortunately state machines have the side effect of limiting reception and transmission rates. That slows performance considerably when there is more traffic to manage. This is likely acceptable for the original use target (home automation), but may not provide adequate performance in other applications. And then, of course, it is questionable if QOS 1 is needed at all as this device should only run on a local network anyway. And use of "batch change" commands can get around a lot of the performance issues.
Thought: I am still struggling with the Home Assistant Header Toggle button ... and wondering if Disconnect/Reconnect would benefit that in some way. Probably not as I think it will be transparent to Home Assistant as it should only be a Mosquitto-to-NetworkModule interaction. But I just re-ran some tests with QOS 0 and find that the Header Toggle button works even worse that way than with QOS 1 enabled. So something else is going on here that hasn't made itself apparent. Maybe disconnect/reconnect would in some way also improve QOS 0 operation? Is it possible that never performing a disconnect/reconnect when sending config messages to Home Assistant is what upsets the HA processes?
Ahh ... more experiments and more testing.

yozik04, 2022-11-16:
No. Please do not do any Disconnect/Reconnect there. HomeAssistant does not expect that. It will receive last_will messages and mark entity as offline. I do not remember if we define last will... But do not do reconnect. It is wrong.

nielsonm236, 2022-11-16:
Hmmm ... OK. But I'm not sure what you say is correct.
First - Yes we do have last will. But from online research "Each client can specify its last will message when it connects to a broker. The last will message is a normal MQTT message with a topic, retained message flag, QoS, and payload. The broker stores the message until it detects that the client has disconnected ungracefully. In response to the ungraceful disconnect, the broker sends the last-will message to all subscribed clients of the last-will message topic. If the client disconnects gracefully with a correct DISCONNECT message, the broker discards the stored LWT message." So, if I properly disconnect no last will should be invoked.
Second - The intention would be to start a connection when we first power up with Connect with Clean Session = True. Thereafter the Reconnect would be with Clean Session = False, which effectively continues the session so there is no loss of in-flight messages (if any).
I can experiment with this in my test bed to see what happens. From online research it appears that frequent Disconnect/Reconnect is typical with edge devices, so I would think Home Assistant would be having all sorts of trouble if it didn't like it. But as I mentioned before Home Assistant shouldn't even see it happen, so I'm not sure why it would be causing the current bizarre behavior. On the other hand, I am unable to explain why the QOS 0 configuration operates so much worse than the QOS 1 configuration given there are no lost or retried messages. But in QOS 1 there ARE PUBACK messages ... does that subtly affect the timing that Home Assistant sees, so it works slightly better? I don't know.

nielsonm236, 2022-11-16:
I should add that our power on process includes generation of all the HA Discovery Config messages. Those would not be repeated with the subsequent Disconnect/Reconnect messages.

nielsonm236, 2022-11-16:
With some further examination I'm not convinced that disconnect/reconnect is a great idea either. Re-SUBSCRIBE is required at reconnect, and any QOS0 messages received after disconnect but before reconnect is established are likely to be lost.

This whole QOS > 0 concept seems to be thoroughly messed up in MQTT. If the messages that need to be resent can only be resent after a disconnect/reconnect (as the spec requires) then there are two problems: 1) If a message is simply lost in the network you likely will never receive it until hours, days, or months later, 2) If you deliberately execute a disconnect/reconnect you expose yourself to lost QOS 0 messages.

So what next in the saga of the Home Assistant Header Toggle? Since I know there is some difference in the behavior of QOS 0 and QOS 1 implementations, and since that difference appears to be timing, perhaps I need to look into that. I could also do a screen-capture-video and send it on the HA community to see if anyone has a clue.

nielsonm236, 2022-11-29:
QOS1 wasn't working the way we expected. See MQTT spec revision 3.1.1 section 4.4.
Closing this issue.

closed 2022-11-29 by nielsonm236