nielsonm236/NetMod-ServerApp #123: Allow short press of Reset button to reboot the module, and a long press to restore defaults

nielsonm236, 2022-11-12:
Someone else once requested this, I think by email but they never entered an Issue (at least I couldn't find one).
The request is to allow a short press of the Reset button to reboot the module without restoring default settings (just like the Reboot button on the Configuration page). A long press of the Reset button would still cause default settings to be restored.

This short press to cause a reboot helps in the situation where the module stops responding preventing access to the Reboot button on the Configuration page. I've found this is mostly an issue when developing new code. There may also be cases where rebooting a module at the hardware location would come in handy if a Browser is not available.

While performing recent testing this would have come in handy, so I'm adding it to the code as follows:
- Inadvertently bumping the button (button press less than 1 second) will have no effect.
- Pressing the button for 1 second, but less than 5 seconds, will result in a Reboot without loss of settings.
- Pressing the button for 5 seconds or longer will result in default settings to be restored.

sjorsBe, 2022-11-26:
Hi mike, 
I did understood that there is very limited space left ( ram and programming space) in the module. 
whenever this idea would use precious space, i would suggest to cancel it. There is an easy way around a reset.  Either through the the web interface or just taking the power off. 
Its up to you, but i would advice to save space for non work around issues 

kind regards 
Sjors

nielsonm236, 2022-11-26:
Thank you for the reply Sjors. I already put this change in ... less than 10 byte Flash impact and no RAM (just a compare on the counter already tracking the 5 second button push). And it has come in handy in test already.
Yes ... you'll see me moan and groan when a feature request is Flash or RAM intensive. But I'll still experiment as I sometimes find a way without much impact. We've packed a lot of stuff in 32K flash / 1536 bytes RAM.

nielsonm236, 2022-11-29:
Addressed in pre-release 20221127 0000

nielsonm236, 2022-12-17:
Addressed in release 20221217 1925.

closed 2022-12-17 by nielsonm236